arduino/arduino-ide #2419: Ctrl+; to add/remove semicolon at end of line

MihaiZecheru, 2024-04-18:
### Describe the request

When the Ctrl+; shortcut is pressed, toggle the semicolon at the end of the line. Add a semicolon if there isn't one, and remove the semicolon if it exists.

### Describe the current behavior

Ctrl+; doesn't do anything currently.

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2025-02-10:
Hi @MihaiZecheru. Thanks for your suggestion. We have decided that the addition of the requested functionality is out of scope for the Arduino IDE project.

It is possible to add additional functionality to Arduino IDE through the tremendous number of 3rd party VS Code extensions that have been created by the open source community. We do not provide official support for any such 3rd party extensions, but after a quick search I found a nice looking one that provides the exact functionality you requested:

https://open-vsx.org/extension/hideoo/trailing

If you would like instructions for installing that extension in Arduino IDE, you can post over on Arduino Forum:

https://forum.arduino.cc/c/development-tools/ide-2-x/93

We'll be happy to provide assistance over there.

closed 2025-02-10 by per1234

MihaiZecheru, 2025-02-10:
I didn't know that vscode extensions worked with Arduino IDE. Thank you!