arduino/arduino-ide #2423: variable ARDUINO has wrong value

mikrocoder, 2024-04-26:
### Describe the problem

The variable ARDUINO returns a completely wrong value of 167 in the IDE 2.3.2. It should actually return 232. In the IDE 1.8.19 it is still correct.

### To reproduce

.

### Expected behavior

The variable ARDUINO returns a completely wrong value of 167 in the IDE 2.3.2. It should actually return 232. In the IDE 1.8.19 it is still correct.

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

11

### Additional context

.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-04-26:
Hi @mikrocoder. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/725.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

closed 2024-04-26 by per1234

mikrocoder, 2024-04-26:
Hi,

the linked thread is displayed as open but is closed. I cannot write there. Because there is more wrong with the variables. In the IDE 2. I can't recognize in the thread what is changed when and how. I don't understand the content of the thread.

```Serial.printf("Used IDF version for compilation: %s\n", esp_get_idf_version());
  Serial.printf("Used ESP arduino platform for compilation: %s\n", getArduinoPlatformVersion());
  Serial.printf("Used IDE version for compilation: %u\n", ARDUINO);
  Serial.printf("GNU C++ version: %s\n", __VERSION__);
  Serial.printf("C++ standard: %ld\n", __cplusplus);
```
shows me
```
Used IDF version for compilation: v4.4.5
Used ESP arduino platform for compilation: 2.0.13
Used IDE version for compilation: 10607
GNU C++ version: 8.4.0
C++ standard: 201103
```

Are you using the original Espressif Core? According to that
[ESP-IDF C++ Support](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html)
C++23 should be set.
A more up-to-date toolchain would have to be available for this. Since I can't find anything in the folder structure of the IDE 2 (I don't see any files of a framework, no libs) I have to ask if the information is correct?

A variable called _runtime.ide.version_ is also not known to IDE 2.

Can you please explain what is correct and what is not?
Thank you very much.

per1234, 2024-04-26:
> I cannot write there

That is fine. There is no need to write there. The problem is already reported. You can either be patient and wait for the developers to make a fix, or you can [submit the fix yourself](https://arduino.github.io/arduino-cli/0.35/CONTRIBUTING/#pull-requests).

>Are you using the original Espressif Core? According to that
[ESP-IDF C++ Support](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html)
C++23 should be set.
A more up-to-date toolchain would have to be available for this.

That subject is off topic for this repository (and for the `arduino/arduino-cli` repository as well). If you would like to discuss the subject, use Arduino Forum:

https://forum.arduino.cc/c/software/ide-2-x/93

> A more up-to-date toolchain would have to be available for this. Since I can't find anything in the folder structure of the IDE 2 (I don't see any files of a framework, no libs) I have to ask if the information is correct?

This issue tracker is only to be used for reporting bugs and making feature requests for the Arduino IDE 2.x code base. We do not provide support here. Please use Arduino Forum for that.

> A variable called runtime.ide.version is also not known to IDE 2.

It is not a variable. It is a platform property. You can learn about it by reading the Arduino Platform Specification:

https://arduino.github.io/arduino-cli/dev/platform-specification/#:~:text=%7B-,runtime.ide.version,-%7D%3A%20the%20version

I would be happy to discuss the subject further if you make a post over on Arduino Forum.