arduino/arduino-ide #2424: Electron 27 is EOL

alucryd, 2024-04-27:
### Describe the request

Hi there, electron 27 has reached end of life on April 16, would it be possible to upgrade to a newer release?

### Describe the current behavior

Electron 27 will soon be removed from that Arch Linux repositories and arduino-ide is currently blocking this removal.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Arch Linux

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

alessandromrc, 2024-05-01:
Best would be going to Electron 30 directly... The IDE is still really outdated on it's own.

alerque, 2025-01-25:
Since the above, even Eletron 30 is how EOL (as it 31). The current version [is 34](https://releases.electronjs.org/) with 3 releases supported at a time. Perhaps just as importantly 27 uses a version of NodeJS that is also EOL and now has reported CVEs. That makes it urgent for distros to try to mitigate, and the most obvious mitigation is to get EOL versions out of package repositories. For Arch Linux the very last reason to keep Electron 27 or NodeJS Hydrogen around at all is only for this package. I don't want to see it become collateral damage, but **please be aware that we can't keep packaging EOL stuff forever**. As an upstream project that wants to stay easily available to end users, please do consider making updating Electron an urgent priority. Thanks.

alessandromrc, 2025-01-25:
> Since the above, even Eletron 30 is how EOL (as it 31). The current version [is 34](https://releases.electronjs.org/) with 3 releases supported at a time. Perhaps just as importantly 27 uses a [...]

From what I remember the main issue back then was Theia not being really updated... I am not aware if things changed recently but I was able to recompile the IDE with newer versions of Electron, there were lot of packages abandoned by their maintainers tho.

Like you said the focus of the IDE should be on upgrading to a newer Electron version but also either finding more "modern" packages that are currently supported or directly rewrite the functionalities of those inside the IDE sourcecode.

alerque, 2025-01-25:
While true and fully maintained dependencies is a worthy thing to work towards, having some unmaintained deps in the SBOM is not going to get the whole app booted from distro packaging the way that using EOL'ed Electron / NodeJS is going to do. Not that it can't also have that effect of course, but the urgency scale is typically different.

oech3, 2025-05-08:
https://gitlab.archlinux.org/archlinux/packaging/packages/arduino-ide/-/issues/2

And nice to add version of electron to `pakage.json`.

oech3, 2025-05-09:
@alessandromrc Is
>I was able to recompile the IDE with newer versions of Electron,

still valid for Electron 31 on Linux and is it able to run it ? And where are `libffmpeg` comming from at build time?

alessandromrc, 2025-05-10:
> @alessandromrc Is
> >I was able to recompile the IDE with newer versions of Electron,
> 
> still valid for Electron 31 on Linux and is it able to run it ? And where are `libffmpeg` comming from at build time?

I really don't remember anything about `libffmpeg`, maybe it's a dependency that has been added later on?

oech3, 2025-05-11:
@alessandromrc I succeed to build with `Electron 31`, but not `34` yet by `C++20` err.
There is a cance to bump the version of Electron of this project by [rejecting ffmpeg coming from nowhere](https://github.com/eclipse-theia/theia/issues/13370).

oech3, 2025-06-13:
https://github.com/eclipse-theia/theia/pull/15837

kjkent, 2025-07-27:
Just an update since @oech3's comment: the referenced PR has been merged and Theia has had three subsequent releases. The latest release bumped Electron again, to `37.2.1`. Would there be anything else precluding Arduino IDE using a more recent Electron version, or is there a lack of resourcing (or interest) to do so?

alessandromrc, 2025-07-28:
> Just an update since [@oech3](https://github.com/oech3)'s comment: the referenced PR has been merged and Theia has had three subsequent releases. The latest release bumped Electron again, to [...]

In my opinion the issue is just related to nobody wanting to spend time to "fix" what is already working fine... Also, from what I remember, I've had a lot of issues updating to any newer Electron versions because of dependencies being heavily deprecated, meaning they would need to be rewritten either as an npm library or internally within the IDE.