arduino/arduino-ide #2426: IDE starts outside of the viewing area

sepp89117, 2024-04-30:
### Describe the problem

If the IDE is used on a second screen in Windows 10 and closed, it will open again on the second screen when you restart it. But if there is no second screen connected the next time you start the IDE, the IDE starts outside of the viewing area and cannot be brought to the first screen.
Using Windows key+arrow keys also has no effect on the IDE window.

### To reproduce

See problem description

### Expected behavior

The IDE should start on the main screen.
The IDE window should be manageable with the Windows key + arrow keys

### Arduino IDE version

2.3.2.0

### Operating system

Windows

### Operating system version

Windows 10 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

dankeboy36, 2024-04-30:
> the IDE starts outside of the viewing area and cannot be brought to the first screen

Please see if this workaround helps to fix it: https://github.com/arduino/arduino-ide/issues/52#issuecomment-1660870738.

alessandromrc, 2024-05-01:
> > the IDE starts outside of the viewing area and cannot be brought to the first screen
> 
> Please see if this workaround helps to fix it: [#52 (comment)](https://github.com/arduino/arduino-ide/issues/52#issuecomment-1660870738).

Not a good way at all, best way would be to get the current available area and see if it is less than the one in the JSON file, if so, reset to a viewable area... eg. center of screen.