arduino/arduino-ide #2427: Some small improvement suggestions

Shumatic, 2024-04-30:
### Describe the request

Hi

Thanks for your effort to provide this IDE
- F1 is well hidden :-), couldn it be in the Tools menu
- Could 'recently used' be in the side bar (under libs ect.)
or stay active, it reset if you save under a new name
- I have use 'Toggle Outline View', and can't get shortcut to work (Ctrl+shift+l)
- 'Toggle Outline View' have an Expand to see structures, could it have the same for variables and functions
- and a sort functions for 'Toggle Outline View'

### Describe the current behavior

N/A

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-04-30:
Hi @Shumatic. Thanks for taking the time to submit these suggestions.

> - F1 is well hidden :-), couldn it be in the Tools menu
> - Could 'recently used' be in the side bar (under libs ect.)
or stay active, it reset if you save under a new name

By design, the command palette is not part of the primary user interface of Arduino IDE so we intentionally don't expose it.

> - I have use 'Toggle Outline View', and can't get shortcut to work (Ctrl+shift+l)

I see there is a collision with another keybinding (tracked at https://github.com/arduino/arduino-ide/issues/1658). As a workaround, please configure a unique shortcut for the "**Toggle Outline View**" command and then use that custom shortcut:

1. Select **File > Advanced > Keyboard Shortcuts** (or **Arduino IDE > Advanced > Keyboard Shortcuts** for macOS users) from the Arduino IDE menus.
   The "**Keyboard Shortcuts**" tab will open in the Arduino IDE window.
1. Type `Toggle Outline View` in the "**Search keybindings**" field of the "**Keyboard Shortcuts**" tab.
1. Hover the mouse pointer over the "**Toggle Outline View**" row in the search results.
   A pencil icon ("Edit Keybinding") will appear at the left side of the row.
1. Click the pencil icon.
   The "**Edit Keybinding for Toggle Outline View**" dialog will open.
1. The input field in the dialog contains a text representation of the keyboard shortcut. Adjust this according to your preferences.
   **ⓘ** Notes about the syntax:
   - For most keys you can simply use the name of the key (i.e., `a`, `3`, `/`, `f2`).
   - Use the `+` separator for key combinations (e.g., `ctrl+s`).
   - Use a space separator for [key sequences](https://github.com/eclipse-theia/theia/tree/master/packages/keymaps#key-sequences) (AKA "[chords](https://code.visualstudio.com/docs/getstarted/keybindings#:~:text=current%20context.-,Chords,-%28two%20separate%20keypress)") (e.g., `ctrl+k ctrl+i`).
1. Click the "**OK**" button in the dialog.
   The dialog will close.
1. Click the **X** icon on the "**Keyboard Shortcuts**" tab.

> - 'Toggle Outline View' have an Expand to see structures, could it have the same for variables and functions
> - and a sort functions for 'Toggle Outline View'

The Outline view is provided by the [**Eclipse Theia** IDE framework](https://theia-ide.org/) / [**VS Code**](https://code.visualstudio.com/) on which Arduino IDE is built. We don't have any involvement in the development of this feature. We are satisfied with the current design and functionality of the Outline view.

Although we don't provide any official support for their use, it is possible to extend the capabilities of Arduino IDE by installing VS Code extensions. Thousands of these free open source extensions are available and I see that includes some enhanced versions of the Outline view. If you would like to learn how to install extensions, you can follow the instructions [**here**](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#3rd-party-themes) (although those instructions are framed for installing a theme, themes are just a specific type of VS Code extension, so the same procedure works for any extension). If you want assistance/discussion regarding installing or using extensions with Arduino IDE, post over on the Arduino Forum, and the community will help there:

https://forum.arduino.cc/c/software/ide-2-x/93

Since we already have a dedicated issue tracking the problem of keyboard shortcut collisions and the other suggestions made here are declined, I'll go ahead and close this. 

---

In the future, please submit a separate issue for each distinct subject matter rather than bundling unrelated subjects into a single issue. The reason for this request is that issues that cover multiple distinct subjects are very difficult for us to manage. We might end up losing track of one of the valuable items in such an issue.

Shumatic, 2024-05-01:
I am impressed by your Swift support, thank you.
I will split issues in the future,