arduino/arduino-ide #2429: 2.3.2 AppImage fails to start due to missing sandboxing

AndKe, 2024-05-05:
### Describe the problem

```
./arduino-ide_2.3.2_Linux_64bit.AppImage 
[49662:0505/163801.040968:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_arduinl1RTCc/chrome-sandbox is owned by root and has mode 4755.
Trace/breakpoint trap (core dumped)

```

### To reproduce

Start .AppImage on Ubuntu 24.04

### Expected behavior

Work as normal.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

mariovaldez, 2024-05-08:
The issue is with the AppArmor configuration in Ubuntu 24.04, not the AppImage. The change in the configuration is explained in the release notes of Ubuntu 24.04 (security reasons).

Because this problem is caused by the OS configuration, I'm not use what the Arduino IDE team can do, except for documenting the installation procedure.

You can disable the sandboxing restriction for all program with:
```
sudo sysctl -w kernel.apparmor_restrict_unprivileged_userns=1
```
or by adding to /etc/sysctl.d/local.conf:
```
kernel.apparmor_restrict_unprivileged_userns=0
```
But that defeats the purpose of the new AppArmor restriction in Ubuntu 24.04.

You can also create a new AppArmor profile for the Arduino IDE (that allows a non-root user to use the sandboxing in a specific application). If you copy the AppImage to /usr/local/bin/arduino (for example), you can create an AppArmor profile with a configuration file, for example, in /etc/apparmor.d/usr.local.bin.arduino, containing:
```
abi <abi/4.0>,
include <tunables/global>
profile arduino /usr/local/bin/arduino flags=(unconfined) {
  userns,
  include if exists <local/arduino>
}
```
and reloading all AppArmor profiles with:
```
sudo service apparmor reload
```
Now you can run the Arduino IDE without the sandboxing error.

You could also run the Arduino IDE with the --no-sandbox option, but that is, in my opinion, a potentially very bad idea.

AndKe, 2024-05-08:
@mariovaldez Thank you for a good explanation/solution.
Is this going to be easier / fixed in future releases? - so far, I have seen this issue with only two .AppImage applications.

The /usr/local/bin/arduino  you refer to, is that a directory, or is arduino the actual arduino.Appmage?
I have a directory for all my AppImages. it would be tempting to make an AppArmor configuration file for all .AppImage files in that dir.

mariovaldez, 2024-05-08:
@AndKe I don't think this will change with later versions of Ubuntu. My understanding is that allowing normal users to use kernel namespaces has been considered a security risk (the kernel namespaces are used to create the sandbox in some AppImage applications). The change in the AppArmor configuration of Ubuntu is a mitigation for that risk.

Regarding the "/usr/local/bin/arduino" path, that's the program name I used. You can use any filename you like (for example, "/usr/local/bin/arduino.AppImage"). I used the "/usr/local/bin" directory just to follow the Linux FHS guidelines but you can install the program anywhere you want (I'm sure lot of people prefer to install in /opt, or even in a per-user ~/opt or ~/bin directory).

Currently I'm using some other AppImage applications (ungoogled-chromium, teams-for-linux, upscayl) that also require their own AppArmor profile.

AndKe, 2024-05-09:
@mariovaldez  Thank you,  sorry for asking a vague question, I did not mean to ask is it will change in future Ubuntu releases, but if the future Arduino IDE  AppImage releases will require this fix in forseeable future. Or if some changes to the IDE will make this issue be a thing of the past.

mariovaldez, 2024-05-13:
@AndKe The Arduino IDE 2 is really a Typescript/Javascript web application running locally with a Chromium engine (as an Electron app, in this case all packaged in an AppImage).

The Chromium engine uses the sandboxing to reduce security risks. There are several ways to implement the sandboxing. On Linux, the Chromium engine currently uses unprivileged kernel  namespaces (in the past other ways have been used but I think they are deprecated now). 

So, as long as the Arduino IDE uses the Chromium engine provided by the Electron framework, and as long as the Chromium engine requires unprivileged namespaces for the sandboxing, and as long as Ubuntu (or any other distribution) restricts the use of the namespaces, the apparmor profile is needed.

AndKe, 2024-05-20:
Yes, unpriviliged is deprecated now, but at least Ubuntu, when *upgraded* to 24.04 is still accepting unprivileged kernel namespaces - while new installations of the same distro enforces this.
This issue will only grow as time goes, also: it is more likely to annoy new Linux users. (due to new installation)

eccentricOrange, 2024-05-20:
@mariovaldez AFAIK, other applications like VS Code also use the Electron+Chromium+NodeJS combination right? How do they get around this issue? I didn't have to muck around with this when installing VS Code in Ubuntu 24.04

(apologies if I seem ignorant of something, I am not a web developer and do not know about JS/TS-related concepts in depth)

eccentricOrange, 2024-05-21:
Got it, thanks for the explanation!

shaw-a, 2024-06-03:
Hi @mariovaldez, all, 

I am using Ubuntu 24.04. I tried to use the AppArmor profile solution,  I like the idea of complying with canonical's security decisions rather than working around them. However after following these steps, I still get the same error as in the original post. 
1. Download arduino-ide_2.3.2_Linux_64bit.AppImage
1. `sudo mkdir /usr/local/bin/arduino && sudo cp ~/Downloads/arduino-ide_2.3.2_Linux_64bit.AppImage /usr/local/bin/arduino/`
2. `chmod +x /usr/local/bin/arduino/arduino-ide_2.3.2_Linux_64bit.AppImage`
3. create /etc/apparmor.d/usr.local.bin.arduino containing:
```
abi <abi/4.0>,
include <tunables/global>
profile arduino /usr/local/bin/arduino flags=(unconfined) {
  userns,
  include if exists <local/arduino>
} 
```
4. reload AppArmor with `sudo service apparmor reload`

Now, when I run `usr/local/bin/arduino/arduino-ide_2.3.2_Linux_64bit.AppImage` I get the error:

>/usr/local/bin/arduino/arduino-ide_2.3.2_Linux_64bit.AppImage 
[5481:0603/171920.702980:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_arduineWT4bK/chrome-sandbox is owned by root and has mode 4755.
Trace/breakpoint trap (core dumped)

Funnily enough, I am unable to run `arduino` as a command to launch the AppImage even though I expected to since /usr/local/bin is in my $PATH:
```
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```
I'm not sure if this is related to why the solution did not work as expected and I wonder do you have any advice for what to try next.

So far I have tried: 
1. rebooting
2. sudo apt update and sudo apt upgrade, then rebooting
3. setting `PATH=$PATH'
4. double checking all the paths and the contents of the /etc/apparmor.d/usr.local.bin.arduino file
5. downloading the arduino appimage again and repeating the steps

shaw-a, 2024-06-04:
Hi @mariovaldez,

I really appreciate your response and your explanation. Following this advice the solution works for me as expected and I can launch the arduino IDE from anywhere by typing the command `arduino`. It was a big aha moment when you mentioned that it was the file name. Thanks for explaining more about the symbolic links and AppArmor profile, I will definitely try those out.

Kind regards!

zolki, 2024-11-10:
Thank you all for the help. My successful set-up with my `/etc/apparmor.d/arduino-ide` file is:
```
abi <abi/4.0>,
include <tunables/global>
profile arduino-ide* /opt/arduino-ide* flags=(unconfined) {
  userns,
  include if exists <local/arduino*>
}
``` 
I set the SUID attribute flag for the `chrome-sandbox` executable, the `4` ahead of `755`, as the message recommended earlier: `4755`, when the app could not run.

My chosen path was `/opt/arduino-ide_2.3.3_Linux_64bit/` of the unzipped software package, containing also the `arduino-ide` executable file.

My desktop entry shortcut file is `arduino-arduinoide.desktop:`
<details><summary>Details</summary>
<p>

[Desktop Entry]
Type=Application
Name=Arduino IDE
GenericName=Arduino IDE
Comment=Open-source electronics prototyping platform
Exec=/opt/arduino-ide_2.3.3_Linux_64bit/arduino-ide
Icon=arduino-arduinoide
Terminal=false
Categories=Development;IDE;Electronics;
MimeType=text/x-arduino;
Keywords=embedded electronics;electronics;avr;microcontroller;
StartupWMClass=processing-app-Base

</p>
</details>

martin3000, 2024-11-13:
> You could also run the Arduino IDE with the --no-sandbox option, but that is, in my opinion, a potentially very bad idea.

Why should running an appimage without sandboxing be worse than running any other linux binary (without sudo)?

truher, 2025-01-16:
Finally, *this* is the year of Linux on the desktop!

gneiss15, 2025-04-24:
I use another aproach: replace the chrome-sandbox binary inside the AppImage whith a link to the version of the local system.
e.g:

> cd squashfs-root/
rm chrome-sandbox
ln -s $(find /opt -name "chrome-sandbox") chrome-sandbox
cd ..

That wors for me (my installation of Ubuntu 24.).
I am not shure if chrome-sandbox is always installed in /opt, if so, it may be possible to do this inside AppRun.

@per1234 : May You look at this ?

MaKaNu, 2025-04-28:
I came across this issue and just interpreted the error message and updated the permissions to 4755. Got sceptical about that "stickybit" and researched it.

Found an explanation about `sudo chmod 4755` and understand why I have never heard about it before.

Now my question: why does the error point out the stickybit solution and why does the appArmor solution seems to be the better/preferred solution?

gneiss15, 2025-04-28:
The sticky bit and the owner (root) get's lost when the sqashfs inside the Appimage was extracted.
So a call of the chrom-sandbox (from the Appimage) falls. That's why I suggest the use of the local file

Van-too, 2025-06-04:
An example of one solution.
Create a text file named **arduino.sh** in the folder with your appimage file.
Write the following in it:
./**arduino-ide_2.3.6_Linux_64bit.AppImage** --no-sandbox
Save it, open the properties of the file arduino.sh, make the file executable, and run it.

**arduino-ide_2.3.6_Linux_64bit.AppImage** is the name of your file; it may be different.

hechaotudou, 2025-06-20:
sudo  ./arduino-ide_2.3.6_Linux_64bit.AppImage  --no-sandbox
Arduino IDE 2.3.6
Checking for frontend application configuration customizations. Module path: /tmp/.mount_arduin1tFPCu/resources/app/lib/backend/electron-main.js, destination 'package.json': /tmp/.mount_arduin1tFPCu/resources/app/package.json
[12786:0621/044429.782111:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[12786:0621/044429.782151:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[12786:0621/044429.782159:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[12786:0621/044429.782165:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.6" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.2.0" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2025-04-09T11:24:04.835Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{},"showWindowEarly":true,"splashScreenOptions":{},"uriScheme":"arduino-ide"},"defaultLocale":"","validatePreferencesSchema":false,"reloadOnReconnect":true,"uriScheme":"theia","preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.6","cliVersion":"1.2.0","buildDate":"2025-04-09T11:24:04.835Z"}
Starting backend process. PID: 12826
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[12819:0621/044430.072806:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[12819:0621/044430.072889:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[12819:0621/044430.072946:ERROR:gl_display.cc(786)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
Authorization required, but no authorization protocol specified

[12819:0621/044430.073013:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[12819:0621/044430.073029:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[12819:0621/044430.073042:ERROR:gl_display.cc(786)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[12819:0621/044430.073055:ERROR:gl_display.cc(820)] Initialization of all EGL display types failed.
[12819:0621/044430.073069:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[12786:0621/044430.131410:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

[12819:0621/044430.166201:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[12819:0621/044430.166241:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[12819:0621/044430.166256:ERROR:gl_display.cc(786)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
Authorization required, but no authorization protocol specified

[12819:0621/044430.166312:ERROR:angle_platform_impl.cc(44)] Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1086 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[12819:0621/044430.166326:ERROR:gl_display.cc(515)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[... cut ...]
2025-06-20T20:44:30.642Z config INFO Mapped the CLI configuration: {"dataDirUri":"file:///root/.arduino15","sketchDirUri":"file:///root/Arduino","additionalUrls":[],"network":"none","locale":"en"}
2025-06-20T20:44:30.642Z config INFO Validating the CLI configuration...
2025-06-20T20:44:30.644Z config INFO The CLI config is valid.
2025-06-20T20:44:30.644Z config INFO <<< Initialized the CLI configuration.
2025-06-20T20:44:31.101Z root INFO Deploy plugins list: 683.1 ms [Finished 1.148 s after backend start]

TimoHelm, 2025-09-21:
Tried to reproduce the solution but still getting the mentioned "sandbox" error.

My binary in /usr/local/bin is named arduino,

[image]

created file in apparmor.d:

[image]

reloaded apparmor service but still getting error when I try to run arduino:

[image]

Of course I can run the arduino-ide with the `--no-sandbox` option, but I wan´t to avoid this.

What am I missing or where do I make a mistake?
Can anybody help?