arduino/arduino-ide #2430: Hello, I am a user from China

Chenjiek, 2024-05-05:
Hello, I am a user from China and I have found that Arduino cannot support Chinese paths, file names, and variable names. There are 2 billion people in the world who use Chinese. When will Arduino be able to support Chinese paths, file names, and variable names?

per1234, 2024-05-05:
Hi @Chenjiek. Thanks for your report.

> Arduino cannot support Chinese paths, file names

It will not be possible for us to investigate the problem based on such a vague description. I need you to provide the following information:

- Detailed instructions we can follow to reproduce the fault
- Detailed description of the fault that occurs when you perform those instructions, including the full and exact text of any error or warning messages you encounter.

> and variable names?

Variable names are handled by a standard C++ compiler, and thus limits on them are those imposed by the compiler, not by Arduino IDE. You can learn about the subject here:

https://en.cppreference.com/w/cpp/language/identifiers

Note that the compiler may impose even more narrow limitations than what is permitted by the C++ specification, per this note on the page linked above:

> Note: Support of Unicode identifiers is limited in most implementations, e.g. [gcc (until 10)](https://gcc.gnu.org/wiki/FAQ#What_is_the_status_of_adding_the_UTF-8_support_for_identifier_names_in_GCC.3F)

matu6968, 2024-09-13:
> Hi @Chenjiek. Thanks for your report.
> 
> > Arduino cannot support Chinese paths, file names
> 
> It will not be possible for us to investigate the problem based on such a vague description. 

I have tested chinese characters (by making a folder with chinese characters and a file with normal characters aka  测试/main.ino which means test/main.c), opening a sketch in it on a latest snapshot build and it seems to work fine, but once the file name has chinese characters like 测试/测试.ino aka test/test.ino it shows a invalid sketch name error like this
[image]

and you can't proceed unless you rename the file to something that has normal characters.

VS Code (aka the forked part of the IDE) supports chinese characters with no problem reading and saving to them.

per1234, 2024-09-14:
This is expected. The sketch must comply with the specification:

https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-folders-and-files

closed 2024-09-14 by per1234