arduino/arduino-ide #2437: Arduino-Ide Linux failed to start  with Fedora 40 new mesa-driver

Champang, 2024-05-18:
### Describe the problem

Impossible to start Arduino-Ide . Freeze in display
https://forums.fedora-fr.org/d/74794-fedora-40-failed-to-start-arduino-ide

### To reproduce

On terminal :
./arduino-ide

### Expected behavior

kill arduino-ide–gpu-
and 
dnf downgrade mesa*

### Arduino IDE version

arduino-ide_2.3.2_Linux_64bit

### Operating system

Linux

### Operating system version

Fedora 40 Kernel Linux 6.8.9-300.fc40.x86_64 x86_64

### Additional context

Mate Desktop
https://forums.fedora-fr.org/d/74794-fedora-40-failed-to-start-arduino-ide/3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Champang, 2024-05-25:
Version mesa*-24.0.8-1 solve

MythicByte, 2024-07-15:
there a quick fix that is temporary i think, remove the gpuache and then it is working 
sudo rm -r ~/.config/App_name/GPUCache replace App_name with arduino