arduino/arduino-ide #2438: Allow configuring board for IDE to select by default when sketch is opened

matsekberg, 2024-05-20:
Enable the use of some meta-comment in the code so that the appropriate board is automatically selected for that code.

Tsessebe, 2024-06-18:
If I may add my 10 cents...

As I software developer by day; and Arduino enthusiast by night; may I make a suggestion:
**Why not create a config folder (Similar to Visual Studio or JetBrains IDEs), in the sketch folder, where the settings are recoded?**

The config/ meta data can them be sodred in either a JSON file or YAML file. My understanding is that the Arduino-CLI makes use of a YAML file.

Visual Studio creates a .vs folder, where the meta data for the project is stored, similarly JetBrains IDE's creates a .idea folder.
This is a standard practice in Modern IDEs.

This mechanism is also used in text editors like VS Code (.vscode folder).

It is a simple yet elegant way of recording the IDE metadata as well as user selected settings.

stef-ladefense, 2024-08-11:
Hello,

I agree with you, it is missing to have the configuration saved in a file in the sketch directory.

it would be enough for the IDE to save the options as the user changes them and reloads them at the same time as the .ino.

it is a big plus when using several cards with as many projects, that it is configured as it was used at the last compilation.
and to avoid transmitting the configuration in the .ino, a hidden folder with the configuration in it is one of the solutions.