arduino/arduino-ide #2446: Arduino always open the same file at startup

Supertolek, 2024-06-01:
I have the latest version of arduino, and i alreeady tried to uninstall and reinstall it.

I don't know for which reason, but the arduino IDE open the exact same file everytime i launch it. (it was still the same file after having reinstalled it)
I don't know if it is a bug or a feature, but it's very frustrating.

PS: I finally fixed it by removing the path in AppData\Roaming\arduino-ide\config.json , but it's still a thing that some people are encountering. And I don't want to have to do this everytime I leave the IDE.

per1234, 2024-06-01:
Hi @Supertolek. Arduino IDE is designed to open the saved sketches that were open at the time you exited the IDE previously.

Please try this experiment and then report back here with the results:

1. Start Arduino IDE.
1. Select **File > New Sketch** from the Arduino IDE menus.
   A new sketch will open in a new Arduino IDE window.
1. Close all other Arduino IDE windows that are open.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch with the name `Issue2446`.
   **ⓘ** This is just an arbitrary name I chose for the sake of the experiment, without any special technical significance.
1. Select **File > Quit** from the Arduino IDE menus.
   The Arduino IDE window will close.
1. Start Arduino IDE.

Now add a comment here on the issue thread to tell me whether the "**Issue2446**" sketch opened in Arduino IDE at step (7) when you performed that experiment.

Supertolek, 2024-06-01:
Yes, it did. But how can I do so that no files open on startup?

per1234, 2024-06-02:
> Yes, it did.

OK, great. So the IDE is working as intended.

Something to note is that the behavior can be a bit confusing when you are working with example sketches. The reason is that, in order to prevent confusion that could be caused by the user accidentally making modifications to the source examples, Arduino IDE opens example sketches as new sketches containing the example sketch code. So if you have an example sketch open in Arduino IDE and then exit the IDE, you might expect that the example will open the next time you start the IDE, but instead whatever saved sketch you had open at a previous exit will be restored instead.

---

> how can I do so that no files open on startup?

Arduino IDE doesn't have that capability. We are tracking the feature request at https://github.com/arduino/arduino-ide/issues/1532. You can subscribe to that issue if you want to get notifications of any new developments related to that feature.

If you would like to discuss the subject, there is a recent discussion on Arduino Forum that contains some information you might find interesting:

https://forum.arduino.cc/t/new-sketch-when-opening/1259890

You are welcome to join in that forum discussion.

---

Since we have determined that the IDE has the intended sketch restoration behavior, and since we have a previous issue tracking the feature request of an option to disable that behavior, I will close this issue.

closed 2024-06-02 by per1234