arduino/arduino-ide #2460: Unable to load sketches using Arduino as ISP

kalymos, 2024-06-22:
### Describe the problem

It is completely impossible to load a sketch using Arduino as ISP as a programmer, both using MiniCorps and official board definitions, however I have no problem burning bootloaders.
All these manipulations were possible without problem on versions 1.8.19

### To reproduce

After the standard creation of an [Arduino as ISP](https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP/)
selecting the correct target board
choosing the programmer as Arduino as ISP

### Expected behavior

I get these two possible results if I use the standard or MiniCor board definitions.

standard
[image]
[image]

MiniCor
[image]
[image]

### Arduino IDE version

2.3.3-nightly-20240622; 2.3.2

### Operating system

Linux

### Operating system version

opensuse 15.6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-06-22:
Hi @kalymos. In order to upload a sketch using any ISP programmer, you must select **Sketch > Upload Using Programmer** from the Arduino IDE menus rather than using **Sketch > Upload** or clicking the "Upload" button as you would do when uploading a sketch to a board normally.

The reason you didn't have to do that in Arduino IDE 1.x is because it has a feature where the board definition can be configured so that an "Upload Using Programmer" operation is triggered when the user selects **Sketch > Upload** from the menus or clicks the "Upload" button. The **MiniCore** platform is configured to have that behavior when you select **Tools > Bootloader > No bootloader** from the Arduino IDE menus.

We already have an issue tracking the missing "default to Upload Using Programmer" feature: https://github.com/arduino/arduino-ide/issues/103. So I'll close this issue as a duplicate of that one.

If you still have problems uploading after using **Sketch > Upload Using Programmer**, you are welcome to post on Arduino Forum to request assistance. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/avrdude-stk500-bootloader-issues/81

closed 2024-06-22 by per1234

kalymos, 2024-06-22:
Thank you for your quick response.
Your answer is 100% functional, sorry I had not searched for the problem with the right keywords.