arduino/arduino-ide #2462: Slow build process with multiple libraries

niiiksh, 2024-06-23:
### Describe the problem

Build is very slow. Compilation also doesn't start again right after failed compilation.

### To reproduce

Use any big project with multiple libraries on 2-core laptop PC. Compilation is slow like forever. Clangd server slow down everything too. It is the most buggy thing I ever seen.

### Expected behavior

Clangd and build should not be slow.

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-31:
Thanks for taking the time to submit an issue @niiiksh. We have and will continue to make efforts to improve the performance of Arduino IDE and its helper tools as possible, but we don't have a magic wand we can just wave to make the sketch build instantly.

This issue doesn't highlight a specific technical defect that can be addressed so I'll close it as not actionable. If you discover specific technical defects you are welcome to submit detailed formal bug reports for the developers to fix.

closed 2024-07-31 by per1234

niiiksh, 2024-08-01:
@per1234 so, make a magic wand. because downloading index seems to build something too. Because of it starting IDE to fully functional state takes long time.

per1234, 2024-08-01:
We are tracking the issue of exceptional index download process times here: https://github.com/arduino/arduino-cli/issues/2347