arduino/arduino-ide #2465: Way to get rid of annoying variable info pane

j-eyking, 2024-06-29:
### Describe the request

Hovering a function, variable, or whatsoever in the IDE results in a info pane, blocking code. Since it's also of little or no use I would like to get a way to get rid of it. I was looking in the preferences but to my surprise this is a very stripped down section. The list of keyboard shortcuts however is very extensive. But there [this problem](https://github.com/arduino/arduino-ide/issues/2464) appears. So it's hard to find the right shortcut, provided that it exists.

### Describe the current behavior

IDE is showing information about functions and variables in a way it's blocking other code. It seems like it's not possible to switch this of.

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-06-29:
This feature is called the "Hover". It can be disabled in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings):

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar:
   [image]
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Preferences**" tab will open in the Arduino IDE main panel.
1. Type `editor.hover.enabled` in the "**Search Settings**" field of the "**Preferences**" tab.
1. Uncheck the box under the "**Editor › Hover: Enabled**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

@j-eyking please use Arduino Forum to ask questions about using Arduino IDE. This issue tracker should be used to submit formal bug reports or feature requests. Discussions and support requests are always welcome over on Arduino Forum:

https://forum.arduino.cc/c/software/ide-2-x/93

closed 2024-06-29 by per1234

j-eyking, 2024-06-30:
Thank you.

To my opinion it doesn't make sense that advanced settings are only accessable by a shortcut and not via the menu.