arduino/arduino-ide #2471: How wrong this project goes

streanger, 2024-07-17:
### Describe the problem

First time I was using Arduino platform is was about 10 years ago, in 2014. That time Arduino IDE was like plug & play, with no telemetry around. I got back to electronic stuff some time ago and I am really, really irritated of what I see. This is how it works right now:

### To reproduce

1. If I turn on opensnitch (application firewall) and refuse for Arduino Internet connection, then it will not start at all. Ever.
[image]

2. If we let it to run, load some sketch, turn on opensnitch again, and try to compile our sketch it won't do it. It will hangs forever.

[image]

3. Apart from that there are crazy telemetry stuff like `api.ipify.org` and `ipv4.icanhazip.com` all around. Lets be like VSCode. Why not?

### Expected behavior

1. Arduino IDE starts even if Internet connection is blocked by firewall
2. I can compile my code in the similar situation as in point 1
3. Telemetry is disabled by default or is an option for user

### Arduino IDE version

2.3.0

### Operating system

Linux

### Operating system version

5.15.0-105-generic

### Additional context

Maybe this topic was already discussed, maybe you (as a user) allow it at some point of install/first run, but for me it doesn't matter. It shouldn't be as disturbing, as you can't even compile your code, without sending data of what your IP is. Please tell me that I'm wrong, and things looks different.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

dankeboy36, 2024-07-18:
> 1. Arduino IDE starts even if Internet connection is blocked by firewall

https://github.com/arduino/arduino-ide/issues/784 and https://github.com/arduino/arduino-ide/issues/2042 are probably covering this.

> 3\. Apart from that there are crazy telemetry stuff like `api.ipify.org` and `ipv4.icanhazip.com` all around.

It's not telemetry. A 3rd-party lib is checking if IDE is connected to the Internet: https://github.com/sindresorhus/is-online?tab=readme-ov-file#how-it-works.

streanger, 2024-07-18:
> > 1. Arduino IDE starts even if Internet connection is blocked by firewall
> 
> #784 and #2042 are probably covering this.

@dankeboy36 thanks for explanation. They mention that it is related to first run after fresh install. I am using .AppImage file and it happens all the time

> > 3. Apart from that there are crazy telemetry stuff like `api.ipify.org` and `ipv4.icanhazip.com` all around.
> 
> It's not telemetry. A 3rd-party lib is checking if IDE is connected to the Internet: https://github.com/sindresorhus/is-online?tab=readme-ov-file#how-it-works.

good to know

---

apart from that I still don't know why we are blocked at compilation level. I'll wait for some comments in that field

dankeboy36, 2024-07-18:
In Arduino IDE 1.x, the [Arduino AVR](https://github.com/arduino/ArduinoCore-avr) core was included with the IDE. You could simply download the IDE, connect your UNO board, verify your sketch, and the IDE would upload it to your board. It worked because all the necessary Arduino tools and cpp code were part of the IDE.

Recently, Arduino IDE 2.x was introduced with a different approach. It features a simple UI powered by the [Arduino CLI](https://github.com/arduino/arduino-cli). Unlike the previous version, this IDE doesn't require knowledge of where the cores and libraries are stored or how to compile the code and upload via serial. The Arduino CLI handles everything: downloading the core and libraries, installing them, compiling your sketches, and uploading them to the board. While this has many advantages, it also has some drawbacks. Notably, no cores and libraries are included in the IDE distribution; instead, the CLI downloads all the necessary tools and code during the first startup. This is how it works whether you prefer it or not. For more details, you can refer to https://github.com/arduino/arduino-ide/issues/122.

> I am using .AppImage file and it happens all the time

If the first startup fails, subsequent startups are likely to fail as well. 

In case you are using a proxy, there is a workaround available at https://github.com/arduino/arduino-ide/issues/1692#issuecomment-1319560003.

You can also check out these related proxy issues in the Arduino IDE:
 - https://github.com/arduino/arduino-ide/issues/2312
 - https://github.com/arduino/arduino-ide/issues/1438
 - https://github.com/arduino/arduino-ide/issues/1692

Please consider opening only one issue for one problem to simplify the tracking process. If you have additional questions, the Arduino forum is a beneficial resource: https://forum.arduino.cc/c/software/ide-2-x/93

I hope you find a solution for your startup issue and enjoy using the Arduino IDE.

streanger, 2024-07-18:
I'm not quite happy with this situation, but I have nothing more than accept it. Thanks for your attention, mate!

closed 2024-07-18 by streanger

per1234, 2024-07-19:
> I am using .AppImage file and it happens all the time

I suspect it is because Little Snitch is blocking communication via ["localhost"](https://en.wikipedia.org/wiki/Localhost). Arduino IDE consists of multiple independent components and helper tools. Unlike Arduino IDE 1.x, which exclusively uses command line interfaces for communication between the IDE and its tools, Arduino IDE 2.x does communication between some of these via an internal TCP/IP port in your computer.

Although Internet access is not mandatory once the additional tools have been installed on the first run, when the localhost communication is not possible, the IDE can not function.