arduino/arduino-ide #2474: Arduino IDE unusable when compiling for ESP()

iw2fvo, 2024-07-20:
### Describe the problem

Compiling a simple sketch using IDE 2.3.2 for the ESP32 or any of the ESP() family takes a sideral time and so IDE is not usable for practical use

### To reproduce

just compile for ESP() and mesure the elapsed time.

### Expected behavior

problem solution 

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

win_10

### Additional context

Intel CPU type i_5 running at 3.2GB/sec, 1TB SSD, 16 GB_RAM, Windows 10 Pro.
Arduino IDE 2.3.2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-20:
Hi @iw2fvo. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. Many thousands of people, including myself use Arduino IDE with ESP32 boards daily without experiencing exceptional compilation durations. So the problem you are experiencing is something to do with your system rather than the Arduino IDE codebase. For this reason, this topic is more appropriate for the Arduino Forum, where we support users in identifying problems with their systems. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

clovisf, 2024-07-22:
I have the same IDE version and am experiencing the same long delays for compilation to finish. W10 22H2

iw2fvo, 2024-07-22:
It is consolidate common problem. We really hope someone will correct it .
At the moment IDE is not usable for compiling ESP() systems  Too many people is saying this truth ! there is no reason to mask it .
Regards