arduino/arduino-ide #2476: SeeedStudio Xiao ESP32-C6 not being selected the first time via "Tools > board > "

clovisf, 2024-07-22:
### Describe the problem

I recently bought a couple of SeeedStudio's Xiao ESP32-C6. Using the Arduino IDE offline version 2.3.2 I put the boards URL (https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json) on Additional boards manager. 

When first selecting the board via  "Tools > Board > ESP32 > XIAO_ESP32C6" and selecting its COM port on top of the screen, I hit compile and get this error: 
`A fatal error occurred: Could not open COM12, the port doesn't exist
Failed uploading: uploading error: exit status 2`

I then re-select the COM port, which offers me the one last used before the -C6.  I then have to go to "Tools > Board" and select the XIAO_ESP32C again, then the COM port again. And finally the compilation and upload works.

### To reproduce

Get Arduino IDE 2.3.2, put this URL in additional boads manager (https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json). Get a SeeedStudio Xiao ESP32-C6, select it via "Tools > Boards", select its COM port and hit compile/upload.

### Expected behavior

After selecting the Xiao ESP32-C via boards manager and selecting its COM port, I expect the code to be compiled and uploaded without issues (given the code compiles correctly).

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

W10 Home single language 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-07-31:
Thanks for your report @clovisf.

Unfortunately I was not able to reproduce the fault despite quite some trying.

The tricky thing is that **esptool** throws this "`the port doesn't exist`" error under a variety of conditions, including but not limited to the obvious one of the port actually not existing. Of all the conditions, I'm not aware of any one that would be caused by Arduino IDE instead of some external problem (e.g., the port of the board is open in some external application).

Are you able to reproduce the fault consistently, or did it only occur once?

If you are able to reproduce the fault consistently, please provide as detailed instructions as possible I can follow to reproduce it and I'll investigate further.

clovisf, 2024-07-31:
One thing I don't think I mentioned is that when this happen and you go to "Tools > Port" it show the correct port where the Xiao ESP32-C6 is but with a different text to it (something like "COMxx (ESP32 family device)". I then try to select another board (just for seeing what happen) and come back to selecting Xiao ESP32-C6 and the "Tools > Port" is still "COMxx (ESP32 family device)". 

I have been doing two things in order to try and make it work but cannot walk you through the steps: 
1) Erase the Xiao ESP32-C6 with a software called "Flash download tool"
2) In the Arduino IDE I alternate between "Enabled" and "disabled" on "Tools > USB CDC on boot"

Don't know if it helps but I have been using the same setup with Arduino UNO, ESP12 (ESP8266MOD), NodeMCU and DFRobot Beetle ESP32-C3 and it does not happen to any of those.

per1234, 2024-07-31:
Thanks for the additional information. Based on this, I can see that the problem is not caused by anything in the Arduino IDE codebase so I will close this as off topic.

You are welcome to request assistance with the problem over on Arduino Forum:

https://forum.arduino.cc/

>  it show the correct port where the Xiao ESP32-C6 is but with a different text to it (something like "COMxx (ESP32 family device)"

This is normal and expected. The "**esp32**" boards platform developers intentionally configured the platform to do this: https://github.com/espressif/arduino-esp32/pull/9717

closed 2024-07-31 by per1234

clovisf, 2024-08-06:
Sorry but I don't get how this solves the problem I am having. Closing the topic does not help anyone that eventually wants to help me, or even to solve a bug.

per1234, 2024-08-06:
Hi @clovisf.

> Closing the topic does not help anyone that eventually wants to help me

You misunderstood the purpose of this GitHub issue tracker. This is not, never has been, and never will be a place to get help. This is a place to contribute to the development of the code base. You come here to report a bug in the code base. If the bug is verified, at some point in the distant future after that time Arduino will find resources to allocate to assign a developer to fix the bug in the code base and that fix will eventually be distributed to the users via a new release of the project, which might be after months, years, or decades.

As I already explained, the appropriate place to request help is Arduino Forum:

https://forum.arduino.cc/

You will find that volunteer helpers from around the world are standing by on the forum to help 24/7/365 and will respond to any high quality support request within a matter of minutes, quickly getting you back to progressing with your project.