arduino/arduino-ide #2483: over the air programming for ESP32: espota needs --auth= argument to be quoted

jamesalawrence, 2024-08-07:
It is possible to perform an OTA update successfully to an ESP32 device from the command line using a command like
<code>
python3 espota.py -r -i <ipaddr> -p 3232 --auth="This is a valid auth phrase with spaces" -f project.ino.bin
</code>

However if you try to enter "This is a valid auth phrase with spaces", either with or without the quotes into the dialogue box for the OTA password within the Arduino IDE then it fails complaining that "is a valid auth phrase with spaces" are not valid arguments for espota.

Please quote the password that is collected within the dialogue box so that spaces do not cause failure.

per1234, 2024-08-07:
Hi @jamesalawrence. Thanks for your report.

The problem is caused by insufficient quoting in the command template defined in the `platform.txt` configuration file of the "esp32" boards platform, not by a problem in the Arduino IDE codebase. So I will close this issue as off topic for this repository.

I have also submitted a pull request to fix the bug in the "esp32" boards platform configuration so the resolution is in progress even though it isn't appropriate to track it via this issue tracker:

https://github.com/espressif/arduino-esp32/pull/10126

closed 2024-08-07 by per1234

jamesalawrence, 2024-08-07:
Thanks @per1234 for your very speedy response.