arduino/arduino-ide #2484: IDE completely non-functional, CLI works

Violettica, 2024-08-08:
### Describe the problem

The Arduino IDE appears to not be able to use its Arduino CLI backend properly at all, yet using the CLI directly doing equivalent operations appears to get results.

### To reproduce

1. Start a fresh install of Arduino IDE which updates the index. Eq. CLI: `arduino-cli update`
2. Do nothing and wait for the `Arduino_Builtin` library to install. Eq. CLI: `arduino-cli lib install Arduino_Builtin`
3. Install the Arduino AVR board core. Eq. CLI: `arduino-cli core install arduino:avr`

### Expected behavior

In every step of the reproduction, the IDE fails to do these operations and it's required to stop the IDE process and run the CLI equivalents marked before starting the IDE again to make progress.
Symptoms shown are as follows, corresponding to each step:
1. On start-up, hangs forever. Logs show "network unreachable" error. CLI seems to be able to reach the network just fine, though, so logically there is no problem with my network.
2. Arduino_Builtin installation error shown. Network unreachable again.
3. AVR installer cannot reach network. Network unreachable.

### Arduino IDE version

2.3.3-nightly-20240630

### Operating system

Linux

### Operating system version

Garuda Linux Hyprland Edition

### Additional context

CLI testing done with 1.0.3-1 from Arch Linux extra repo.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-08-08:
Hi @Violettica. Thanks for your report.

> Logs show "network unreachable" error.

Please add a comment here that provides the Arduino IDE logs.

You are welcome to collect them via whatever procedure is most convenient. I'll provide instructions for collecting them from the terminal:

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Open [a command line terminal](https://ubuntu.com/tutorials/command-line-for-beginners) in the folder that contains the Arduino IDE installation.
1. Invoke the Arduino IDE executable.
   For example, if you are using the "AppImage" package, the invocation would look like this:
   ```text
   ./arduino-ide_<version>_Linux_64bit.AppImage
   ```
   (where `<version>` is the Arduino IDE version number)
   Or if you are using the ZIP package, the invocation would look like this:
   ```text
   ./arduino-ide
   ```
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Use the mouse to select all the text in the terminal window. Make sure to scroll all the way up to the top.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**C**</kbd> keyboard shortcut to copy the selected text to the clipboard.
1. Paste (<kbd>**Ctrl**</kbd>+<kbd>**V**</kbd>) the copied output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

Violettica, 2024-08-08:
Sure thing. Here's what the logs look like on nightly after deleting `.arduino15` and `.arduinoIDE`, I believe this creates a "fresh install" environment.
```
Arduino IDE 2.3.3-nightly-20240630
Checking for frontend application configuration customizations. Module path: /tmp/.mount_arduinnResnJ/resources/app/lib/backend/electron-main.js, destination 'package.json': /tmp/.mount_arduinnResnJ/resources/app/package.json
[24735:0808/083238.163947:ERROR:object_proxy.cc(577)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.InvalidArgs: No such interface “org.freedesktop.portal.FileChooser”
[24735:0808/083238.164040:ERROR:select_file_dialog_linux_portal.cc(280)] Failed to read portal version property
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.3-nightly-20240630" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "0.36.0-rc.1" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-06-30T03:08:59.213Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.3-nightly-20240630","cliVersion":"0.36.0-rc.1","buildDate":"2024-06-30T03:08:59.213Z"}
Starting backend process. PID: 24843
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/violettica/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2024-08-08T15:32:39.516Z root INFO Backend Object.initialize: 9.4 ms [Finished 0.951 s after backend start]
2024-08-08T15:32:39.519Z root INFO Backend Object.configure: 8.2 ms [Finished 0.951 s after backend start]
2024-08-08T15:32:39.519Z root INFO Backend a.onStart: 2.7 ms [Finished 0.951 s after backend start]
2024-08-08T15:32:39.520Z root INFO Backend a.configure: 0.6 ms [Finished 0.988 s after backend start]
2024-08-08T15:32:39.521Z root INFO Backend d.onStart: 0.4 ms [Finished 0.988 s after backend start]
2024-08-08T15:32:39.522Z root INFO Backend d.configure: 0.1 ms [Finished 0.988 s after backend start]
2024-08-08T15:32:39.523Z root INFO Backend d.initialize: 37.1 ms [Finished 0.988 s after backend start]
2024-08-08T15:32:39.523Z root INFO Backend c.onStart: 0.4 ms [Finished 0.988 s after backend start]
2024-08-08T15:32:39.524Z root INFO Backend s.configure: 3.8 ms [Finished 0.992 s after backend start]
2024-08-08T15:32:39.526Z config INFO >>> Initializing CLI configuration...
2024-08-08T15:32:39.526Z root INFO Backend c.initialize: 0.8 ms [Finished 0.993 s after backend start]
2024-08-08T15:32:39.526Z root INFO Backend x.onStart: 1.0 ms [Finished 0.993 s after backend start]
2024-08-08T15:32:39.527Z config INFO Loading CLI configuration from /home/violettica/.arduinoIDE/arduino-cli.yaml...
2024-08-08T15:32:39.527Z daemon INFO Starting daemon from /tmp/.mount_arduinnResnJ/resources/app/lib/backend/resources/arduino-cli...
2024-08-08T15:32:39.527Z root INFO Backend s.initialize: 0.6 ms [Finished 0.994 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend w.onStart: 1.3 ms [Finished 0.995 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend r.configure: 2.0 ms [Finished 0.995 s after backend start]
2024-08-08T15:32:39.528Z discovery-log INFO start
2024-08-08T15:32:39.528Z discovery-log INFO start new deferred
2024-08-08T15:32:39.528Z root INFO Backend d.initialize: 0.5 ms [Finished 0.996 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend y.onStart: 2.5 ms [Finished 0.998 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend r.configure: 2.2 ms [Finished 0.998 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend s.onStart: 0.7 ms [Finished 1.000 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend h.configure: 0.6 ms [Finished 1.000 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend a.configure: 0.1 ms [Finished 1.000 s after backend start]
2024-08-08T15:32:39.528Z root INFO Backend l.configure: 0.1 ms [Finished 1.000 s after backend start]
2024-08-08T15:32:39.531Z root INFO Theia app listening on http://127.0.0.1:44983.
2024-08-08T15:32:39.531Z root INFO Finished starting backend application: 3.2 ms [Finished 1.003 s after backend start]
2024-08-08T15:32:39.572Z root WARN The local plugin referenced by local-dir:/home/violettica/.arduinoIDE/plugins does not exist.
2024-08-08T15:32:39.573Z root WARN The local plugin referenced by local-dir:/home/violettica/.arduinoIDE/extensions does not exist.
2024-08-08T15:32:40.051Z root INFO Resolve plugins list: 524.3 ms [Finished 1.523 s after backend start]
2024-08-08T15:32:40.411Z daemon INFO time="2024-08-08T08:32:40-07:00" level=info msg="Config file not found, using default values"
[... cut ...]
    at v.evaluateErrorStatus (/tmp/.mount_arduinnResnJ/resources/app/lib/backend/main.js:2:654727)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async s.<anonymous> (/tmp/.mount_arduinnResnJ/resources/app/lib/backend/main.js:2:654397)
Closing channel on service path '/services/electron-window'.
Closing channel on service path '/services/ide-updater'.
Stored workspaces roots: 
No sketches were scheduled for deletion.
```
Need logs for the other steps too?

NohWayJose, 2024-08-27:
@per1234 Just curious. Why wouldn't you just 'pipe' stdOut to a text file. That way you'd be protected from limitations of the line buffer count truncating the output - or have I misunderstood?

per1234, 2024-08-27:
> Why wouldn't you just 'pipe' stdOut to a text file

Because from experience I know that, unfortunately, many people don't any familiarity with working directly with file systems and struggle greatly to perform operations that seem trivial to those of us who do have such familiarity, even when provided detailed instructions for doing so.

I have found that the procedure I describe above is much easier for that class of user to accomplish. I also haven't experienced a single case of the line buffer actually posing a problem after providing these instructions to hundreds of Arduino IDE users in the course of my triage and support efforts.