arduino/arduino-ide #2487: Help menu missing search option on macOS

matthewruzzi, 2024-08-10:
### Describe the problem

Normally Mac apps have a Search field in the Help menu that allows searching all menu items. The Arduino IDE doesn't have this. 

### To reproduce

Open the Mac app and click on the Help menu. 
[image]

### Expected behavior

There would be a Search field. 

Screenshot of VS Code Help menu:
[image]

### Arduino IDE version

Version: 2.3.3-nightly-20240423 Date: 2024-04-23T03:11:06.602Z CLI Version: 0.36.0-rc.1  Copyright © 2024 Arduino SA

### Operating system

macOS

### Operating system version

Sonoma 14.6.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-08-10:
Hi @matthewruzzi. Thanks for your suggestion. Although it might be convenient to have such a thing in a menu with many items, there are only ten items in this menu so the user will surely have no difficulty visually scanning through them to find the one they want.

So we have decided that we would prefer to dedicate Arduino's limited development resources to implementing higher impact features and will be satisfied with the current "Help" menu's UI.

closed 2024-08-10 by per1234

matthewruzzi, 2024-08-11:
The search option in the help menu is something that all Mac apps have by default unless it is specifically disabled or overridden, so it isn't something that would have to be created from scratch. Every other app on my Mac has a search option in the help menu. The Arduino IDE actually temporarily has this option while loading and then it is removed once the help menu finishes loading. 

[image]