arduino/arduino-ide #2498: Arduino 2.3.2 hangs at logo screen on openSUSE tumbleweed

NohWayJose, 2024-08-27:
### Describe the problem

Application window renders with the logo in centre but then hangs. 

### To reproduce

Using any method (GUI or shell) invoke `arduino-ide`. 
I tried latest zip, AppImage and overnight zip/AppImage builds

### Expected behavior

I'd expect the ide to open and load the last opened code

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Operating System: openSUSE Tumbleweed 20240823 KDE Plasma Version: 6.1.4 KDE Frameworks Version: 6.5.0 Qt Version: 6.7.2 Kernel Version: 6.10.5-1-default (64-bit) Graphics Platform: Wayland Processors: 32 × 13th Gen Intel® Core™ i9-13900K Memory: 62.6 GiB of RAM Graphics Processor: AMD Radeon RX 6650 XT Manufacturer: Gigabyte Technology Co., Ltd. Product Name: Z790 AORUS ELITE AX

### Additional context

The following is from bash stdOut - it's probably truncated by the line buffer size (attached file seems different but it's stdOut, piped to a text file (could just be the truncated start of stdOut)

Looks like someone has a similar issue on Fedora in #2458
```
------------------------
// Vertex SKSL
#extension GL_NV_shader_noperspective_interpolation: require
uniform float4 sk_RTAdjust;in float2 position;in float2 localCoord;noperspective out float2 vlocalCoord_S0;void main() {// Primitive Processor QuadPerEdgeAAGeometryProcessor
vlocalCoord_S0 = localCoord;sk_Position = position.xy01;}
// Fragment SKSL
#extension GL_NV_shader_noperspective_interpolation: require
sampler2D uTextureSampler_0_S0;
noperspective in float2 vlocalCoord_S0;void main() {// Stage 0, QuadPerEdgeAAGeometryProcessor
half4 outputColor_S0 = half4(1);float2 texCoord;texCoord = vlocalCoord_S0;outputColor_S0 = (blend_modulate(sample(uTextureSampler_0_S0, texCoord), half4(1)));const half4 outputCoverage_S0 = half4(1);{ // Xfer Processor: Porter Duff
sk_FragColor = outputColor_S0 * outputCoverage_S0;}}
// Vertex GLSL
#version 300 es

#extension GL_NV_shader_noperspective_interpolation : require
precision mediump float;
precision mediump sampler2D;
uniform highp vec4 sk_RTAdjust;
in highp vec2 position;
in highp vec2 localCoord;
noperspective out highp vec2 vlocalCoord_S0;
void main() {
    vlocalCoord_S0 = localCoord;
    gl_Position = vec4(position, 0.0, 1.0);
    gl_Position = vec4(gl_Position.xy * sk_RTAdjust.xz + gl_Position.ww * sk_RTAdjust.yw, 0.0, gl_Position.w);
}

// Fragment GLSL
#version 300 es

#extension GL_NV_shader_noperspective_interpolation : require
precision mediump float;
precision mediump sampler2D;
out mediump vec4 sk_FragColor;
uniform sampler2D uTextureSampler_0_S0;
noperspective in highp vec2 vlocalCoord_S0;
void main() {
    mediump vec4 outputColor_S0 = vec4(1.0);
    highp vec2 texCoord = vlocalCoord_S0;
[... 971 lines cut ...]
noperspective in highp vec2 vlocalCoord_S0;
void main() {
    mediump vec4 outputColor_S0 = vec4(1.0);
    highp vec2 texCoord = vlocalCoord_S0;
    outputColor_S0 = texture(uTextureSampler_0_S0, texCoord, -0.5);
    {
        sk_FragColor = outputColor_S0;
    }
}

Errors:
link failed but did not provide an info log
Storing the sketch as a workspace root: </home/greg/Arduino/BathroomMirrorRelay_sketch_Daylight_saving>.
Skipped storing sketch as workspace root. Already visited: </home/greg/Arduino/BathroomMirrorRelay_sketch_Daylight_saving>.
Stored workspaces roots: /home/greg/Arduino/BathroomMirrorRelay_sketch_Daylight_saving
No sketches were scheduled for deletion.

```
[startup.txt](https://github.com/user-attachments/files/16767631/startup.txt)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

wt, 2025-01-25:
This is also affecting Fedora Rawhide. I am running the stable flatpak version. The `flatpak list` command show this: `Arduino IDE v2  cc.arduino.IDE2 2.3.4   stable  flathub system`

If I manually run with `--disable-gpu` (by running `flatpak run cc.arduino.IDE2 --disable-gpu`), I can see the screen again, but it's super slow.

Dapid, 2025-03-28:
Same here on Fedora 41 on Intel integrated GPUs, but it works as expected in my machine with an Nvidia GPU and drivers.