arduino/arduino-ide #2501: At start of the IDE downloading the 39Mb library_index.json takes minutes to hours.

ednieuw, 2024-09-08:
### Describe the problem

When the IDE is started it start downloading packages. This takes minutes on a fast PC with high speed glasfiber internet and on a older PC it takes forever as I noted at a friend his PC with probably slower internet.
The IDE can not be used properly when the packages are downloaded.

At first I blamed the Espressif esp32 package I had in my Additional boards
 together with the Earl Philpower package_rp2040_index.json 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json was the last line in the Additional boards.

"Downloading package ....". is what is displayed Right under in the IDE window in a annoyingly placed window. (This window blocks the last lines of information) 

"..." is probably the last packages in the additional boards but not what it is downloading and causing the problem

After I removed all additional boards but now I see downloading library_index.json is probably the blame.
This is a json  file of 39MB with 967379 lines of text!!

### To reproduce

Start the IDE for the first time at the start of a new day. 
Probably the file is downloaded two a three times per day but difficult to reproduce
After a successful download the IDE does not download the packages for several hours

### Expected behavior

A fast startup like the IDE1 where there is no noticeable time of installing packages at startup.

### Arduino IDE version

2.3.3 nightly 20240422

### Operating system

Windows

### Operating system version

W11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-09-08:
Hi @ednieuw. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/2347.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2024-09-08 by per1234

vdevan, 2024-09-26:
I downloaded the latest Arduino IDE v2.3.3 today. This problem still exists. I had two files in this order
http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json 
First one gets downloaded second one hangs. I deleted the second one but now the first one hangs
Needs rectified ASAP. Never had this issue in previous versions