arduino/arduino-ide #2502: 2.3.2 not working with Linux Kernel 6.1.0

dsssssssss9, 2024-09-08:
### Describe the problem

recently installed fresh copy of Ubuntu 24.04 on my pc along with all my usual programs

For 1st couple of days arduino IDE worked fine but yesterday for some reason it wont load!

All i get is a pulsating logo

i have wiped / reinstalled  IDE ,, still would not  load .

After couple of days having exhausted the various search results for this problem & nothing worked I was about to give up & stick with IDE v1.

As i stated originally  ide 2.3.2 WAS working the day before so i looked through the logs to see what had changed in the last 24 hours- the only thing i could find was that i had upgraded the kernel from V6.9.1 to V6.10.1
I went back to the previous version ( 6.9.1) & Ide 2.3.2 started normally!!

Can anyone give me a clue as to why ???

I see there are several Kernel version from both 6.9.x & 6.10.x - is this just a problem with 6.10.1 or maybe with entire 6.10.x ??

### To reproduce

Install Kernel 6.1.0 on Ubuntu 24.04 MATE

### Expected behavior

IDE starts up as usual

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Ubuntu  24.04LTS MATE

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-09-09:
Hi @dsssssssss9. I'm going to ask you to provide some additional information that might help us to identify the problem.

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Open [a command line terminal](https://ubuntu.com/tutorials/command-line-for-beginners) in the folder that contains the AppImage file.
1. Type the following command:
   ```text
   ./arduino-ide_<version>_Linux_64bit.AppImage
   ```
   (where `<version>` is the Arduino IDE version number)
1. Press the <kbd>**Enter**</kbd> key.
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Use the mouse to select all the text in the terminal window. Make sure to scroll all the way up to the top.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**C**</kbd> keyboard shortcut to copy the selected text to the clipboard.
1. Paste (<kbd>**Ctrl**</kbd>+<kbd>**V**</kbd>) the copied output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

per1234, 2024-10-02:
From @dsssssssss9's Arduino Forum thread on the subject:

>>```text
>>Warning: terminator_CreateInstance: Failed to CreateInstance in ICD 3.  Skipping ICD.
>>```
>
>ICD is the acronym for "[installable client driver](https://en.wikipedia.org/wiki/OpenCL#:~:text=at%20runtime.%20An-,installable%20client%20driver,-%28ICD%29%20must%20be)". This seems to be mostly often related to the GPU (i.e., the graphics card on your computer). 
>
>I haven't seen previous reports of this specific symptom, but when other users have had problems related to the graphics card, the workaround has been to add the `--disable-gpu` to the Arduino IDE command line invocation.
>
>So if you are still able to reproduce the problem, please try starting Arduino IDE with this command:
>
>```text
>/home/jkd-mini2024/Arduino2024/arduino-ide_2.3.2_Linux_64bit.AppImage --disable-gpu --no-sandbox
>```

@dsssssssss9 did you try that? If so, please add a comment here to let us know the result. If not, please do.

dsssssssss9, 2024-10-02:
> From @dsssssssss9's Arduino Forum thread on the subject:

> 
> @dsssssssss9 did you try that? If so, please add a comment here to let us know the result. If not, please do.

Hi 

Sorry for the lack of replies but I've been away....

No I did not try the disabled -gpu switch yet .

I managed to get ide working by downgrading from  kernel 6.10.1 to a previous 6.9.x version that I have been running since..

I am back at my desk in a couple of days & will upgrade to 6.10.x version & test your suggestion