arduino/arduino-ide #2517: Canceling compilation or upload doesn't kill tool process

per1234, 2024-09-26:
### Describe the problem

When a compile, upload, or burn bootloader operation is triggered by the Arduino IDE user, command line tool(s) (e.g., avr-gcc, AVRDUDE) are invoked. It is these tools that actually perform the compilation, upload, or burn the bootloader.

A "**CANCEL**" button was added to the compile, upload, and burn bootloader progress notifications (https://github.com/arduino/arduino-ide/pull/2334 / https://github.com/arduino/arduino-ide/commit/2a325a5b74a7ea80645cd2da74588f706292f4ff). Clicking this button cancels the operation in the Arduino IDE GUI, but doesn't kill the tool process. The tool process continues to run just as it would have done if the button had not been pressed

🐛 The button is misleading, since the user is sure to have the reasonable expectation that the tool process will be canceled.
🐛 This can cause very confusing problems when the user triggers a new operation and it has a collision with the process that was still running from the canceled operation. Normally the existence and status of an ongoing process would be communicated to the user via the Arduino IDE UI, but after using the "cancel" feature, there is no such communication. So it will be difficult for the user (and those who support their report) to understand why this is happening.

### To reproduce

#### Equipment

- Any board or other hardware that produces a serial port.

#### Demo

1. Select **File > New Sketch** from the Arduino IDE menus.
1. Select the port from Arduino IDE's **Tools > Port** menu.
1. Select **Tools > Board > Arduino AVR Boards > Arduino Duemilanove or Diecimila** from the Arduino IDE menus.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for the "**Uploading...**" notification to appear.
1. Click the "**CANCEL**" button in the notification.
1. Select **Sketch > Upload** from the Arduino IDE menus.

🐛 The upload fails with an unexpected "`Access is denied.`" error. The reason is that the `avrdude` process from the first upload operation that was triggered at step (4) was still running, even though that operation was canceled.

### Expected behavior

The process is killed when the "**CANCEL**" button is clicked.

**- OR -**

If it is not possible to kill the process, then remove the misleading "**CANCEL**" button.

### Arduino IDE version

2.3.3

### Operating system

- Linux
- Windows

### Operating system version

- Ubuntu 24.04
- Windows 11

### Additional context

Related: https://github.com/arduino/arduino-ide/issues/1199

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

cmaglie, 2024-10-24:
Fixed in https://github.com/arduino/arduino-cli/pull/2726

closed 2024-10-24 by cmaglie

kittaakos, 2024-11-14:
> Fixed in [arduino/arduino-cli#2726](https://github.com/arduino/arduino-cli/pull/2726)

Note that the fix is not yet in the nightly builds.

https://github.com/arduino/arduino-ide/blob/63e9dfd7f5949c77ffbfb6175e2e9c2991317bca/arduino-ide-extension/package.json#L172-L174