arduino/arduino-ide #2522: Port selection dialog under "Tools" is greyed out/unselectable

AzyFloof, 2024-09-28:
### Describe the problem

After upgrading to 2.3.3 the Port option under Tools is greyed out. The IDE won't let me select any boards, regardless of which ones I have plugged in.
Uploading sketches obviously fails because of this, because the IDE thinks I don't have a board connected.
Using the Board/Port selection dropdown shows no boards listed

### To reproduce

Plug in any arduino (I tried a bare ATMEGA328 with an FTDI dongle, an Arduino Micro, an Arduino Leonardo, and a Pololo A*Star 328 16MHz).
The ports option under tools is greyed out and I cannot select any board

### Expected behavior

I expect a list of plugged in and supported boards to appear as a list under the Ports sub menu, and under the boards/ports dropdown list

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

10

### Additional context

The issue is not present in 2.3.2 if I downgrade
I've tried uninstalling and reinstalling 2.3.3 manually but the issue persists
The ports of the connected boards show up in the device manager list in Windows
The problem persists if I try different cables and different USB ports
All my drivers are up to date

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-10-05:
Hi @AzyFloof. Thanks for your report. I'm going to ask you to provide some additional information that might help us to identify the problem.

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Start [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1).
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<username>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the <kbd>**Shift**</kbd> key while clicking the right hand button on the mouse.
   A context menu will open.
1. Select "**Open PowerShell window here**" from the menu.
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the <kbd>**Enter**</kbd> key.
   Debug output should now be printed to the PowerShell window as Arduino IDE starts.
1. Wait for Arduino IDE to finish starting up completely.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**A**</kbd> keyboard shortcut.
   This will select all the text in the **PowerShell** window.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**C**</kbd> keyboard shortcut.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (<kbd>**Ctrl**</kbd>+<kbd>**V**</kbd>) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

RicoEllis13, 2024-12-12:
I have similar issue, so, I'm going to write it down here.

RicoEllis13, 2024-12-12:
I have similar issue, so I'm going to write about it down here.
I've installed Nullab AVR Compatible Boards (link: https://raw.githubusercontent.com/nulllaborg/arduino_nulllab/master/package_nulllab_boards_index.json) long time ago and it was working just fine, then at one moment, I wasn't able to choose this board in Tools (board didn't change) and when I tried to choose it in "Select Other Board and Port" window, that window blacked out. Same thing happends if I try to write in search bar "nu".
[image]
I've tried to delete and reinstall Nullab board in board manager, then Arduino IDE, but it didn't fix anything. I've tried other versions and this issue occurs only in 2.3.3v and 2.3.4v (it also occurs in Nightly Build). It doesn't occur on clear PC (where I never had Arduino IDE 2.x) in same circumstances. I have some other boards and libraries installed though. 
Also, here's the thing you asked author for, if that helps:

PS D:\Users\Mi\AppData\Local\Programs\Arduino IDE> & ".\Arduino IDE"
PS D:\Users\Mi\AppData\Local\Programs\Arduino IDE>
Arduino IDE 2.3.3
Checking for frontend application configuration customizations. Module path: D:\Users\Mi\AppData\Local\Programs\Arduino IDE\resources\app\lib\backend\electron-main.js, destination 'package.json': D:\Users\Mi\AppData\Local\Programs\Arduino IDE\resources\app\package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.3" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.0.4" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-09-25T09:41:18.242Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.3","cliVersion":"1.0.4","buildDate":"2024-09-25T09:41:18.242Z"}
Starting backend process. PID: 4532
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///c%3A/Users/Mi/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2024-12-12T19:14:34.034Z root INFO Backend Object.initialize: 4.5 ms [Finished 0.658 s after backend start]
2024-12-12T19:14:34.034Z root INFO Backend Object.configure: 3.5 ms [Finished 0.658 s after backend start]
2024-12-12T19:14:34.034Z root INFO Backend a.onStart: 1.8 ms [Finished 0.658 s after backend start]
2024-12-12T19:14:34.035Z root INFO Backend a.configure: 0.3 ms [Finished 0.681 s after backend start]
2024-12-12T19:14:34.035Z root INFO Backend d.onStart: 0.2 ms [Finished 0.681 s after backend start]
2024-12-12T19:14:34.036Z root INFO Backend d.configure: 0.1 ms [Finished 0.681 s after backend start]
2024-12-12T19:14:34.036Z root INFO Backend d.initialize: 22.5 ms [Finished 0.681 s after backend start]
2024-12-12T19:14:34.037Z root INFO Backend c.onStart: 0.2 ms [Finished 0.681 s after backend start]
2024-12-12T19:14:34.037Z root INFO Backend s.configure: 1.0 ms [Finished 0.682 s after backend start]
2024-12-12T19:14:34.038Z config INFO >>> Initializing CLI configuration...
2024-12-12T19:14:34.038Z root INFO Backend c.initialize: 0.4 ms [Finished 0.682 s after backend start]
2024-12-12T19:14:34.039Z root INFO Backend x.onStart: 0.6 ms [Finished 0.683 s after backend start]
2024-12-12T19:14:34.039Z config INFO Loading CLI configuration from c:\Users\Mi\.arduinoIDE\arduino-cli.yaml...
2024-12-12T19:14:34.040Z daemon INFO Starting daemon from D:\Users\Mi\AppData\Local\Programs\Arduino IDE\resources\app\lib\backend\resources\arduino-cli.exe...
2024-12-12T19:14:34.040Z root INFO Backend s.initialize: 0.3 ms [Finished 0.683 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend w.onStart: 0.8 ms [Finished 0.684 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend r.configure: 1.3 ms [Finished 0.684 s after backend start]
2024-12-12T19:14:34.040Z discovery-log INFO start
2024-12-12T19:14:34.040Z discovery-log INFO start new deferred
2024-12-12T19:14:34.040Z root INFO Backend d.initialize: 0.4 ms [Finished 0.685 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend y.onStart: 1.7 ms [Finished 0.686 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend r.configure: 1.4 ms [Finished 0.686 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend s.onStart: 0.6 ms [Finished 0.687 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend h.configure: 0.5 ms [Finished 0.687 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend a.configure: 0.1 ms [Finished 0.688 s after backend start]
2024-12-12T19:14:34.040Z root INFO Backend l.configure: 0.1 ms [Finished 0.688 s after backend start]
2024-12-12T19:14:34.043Z root INFO Theia app listening on http://127.0.0.1:53078.
[... cut ...]
Storing the sketch as a workspace root: <c:\Users\Mi\Documents\Arduino\Arduino_github\RicoEllis13-wireless_gloves_with_finger_tracking\Программы правых рук\Glove_Right_modded_chAdr1>.
Skipped storing sketch as workspace root. Already visited: <c:\Users\Mi\Documents\Arduino\Arduino_github\RicoEllis13-wireless_gloves_with_finger_tracking\Программы правых рук\Glove_Right_modded_chAdr1>.
Stored workspaces roots: c:\Users\Mi\Documents\Arduino\Arduino_github\RicoEllis13-wireless_gloves_with_finger_tracking\Программы правых рук\Glove_Right_modded_chAdr1
No sketches were scheduled for deletion.

d3lta-v, 2025-01-15:
Hi all, 

I was experiencing this exact issue with my own company's board cores made for Arduino IDE (I am on version 2.3.4, and this issue did not exist on version 2.3.2).

I believe I have determined the source of the problem, as my own debug logs are not too dissimilar to the example above:

```
2025-01-15T06:52:54.768Z root ERROR c: Invalid FQBN: SSTuino II Series Boards:megaavr:4809
    at s (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:9316246)
    at b (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:2055780)
    at a (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:2054726)
    at file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:2054935
    at Array.map (<anonymous>)
    at e.decorateBoards (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:2054825)
    at l.renderBoards (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:1456660)
    at l.renderContainer (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:1456409)
    at l.render (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:1455995)
    at Ta (file:///Applications/Arduino%20IDE.app/Contents/Resources/app/lib/frontend/bundle.js:2:10218422)
```

This indicates that Arduino IDE versions 2.3.3 onwards contains a bug that does not recognise spaces in the `name` field of package.json files, which forms part of the board's FQBN. However, the [latest specification for the `name` field in package.json](https://arduino.github.io/arduino-cli/1.1/package_index_json-specification/#json-index-file-contents) does not indicate that spaces and special characters are prohibited which can be confusing to new developers. 

```
{
  "packages": [
    {
      "name": "nulllab avr compatible boards",               <--------THIS IS THE ISSUE
      "maintainer": "nulllab.org",
      "websiteURL": "https://github.com/nulllaborg/arduino_nulllab",
      "email": "nulljun@nulllab.cn",
...
```

---------------------------------------------

The fastest fix is to temporarily downgrade to IDE 2.3.2, request the board maintainer to modify their package.json's `name` and `packager` fields to not contain any spaces (see [my updated package.json](https://fourierindustries-llp.github.io/SSTuino_II_Core/package_FourierIndustries-LLP_SSTuino_II_Core_index.json) for an example). After which, delete the Additional Board Managers URL, exit the IDE, and add back the URL to force a refresh. Remove the existing board package and reinstall the latest version of the board and the issue should be completely gone.

[image]

^ Delete this URL and add it back to force a refresh 

--------------------------------------

One final note: Arduino IDE 2 has a smart feature which caches the last used board for every sketch so that sketches will open with the right board and serial port upon startup, but this can cause issues if the last used board has this issue with their FQBN. You will need to downgrade the IDE, change the selected board to a known good board with no such issue, install the patch, upgrade the IDE and change the selected board back. If Arduino maintainers knows how to clear this cache manually (and maybe add a button to clear the cache in the settings), let us know! :)

per1234, 2025-04-29:
Thanks for the information. The developers have determined that support for spaces in the vendor name will not be provided. The restrictions on the vendor name have been (😢 poorly) documented:

https://github.com/arduino/arduino-cli/pull/2509

closed 2025-04-29 by per1234