arduino/arduino-ide #2526: Custom partitions.csv on ESP32 is ignored

aIecxs, 2024-10-07:
### Describe the problem

Custom partitions on ESP32 using ESP-IDF (not using Arduino ESP32) does not calculate app memory correctly on Windows Arduino IDE 2.3.3. The partitions.csv file in Sketch Directory is parsed and checked for proper alignment, but uploading app failes with Sketch too big;

(logfile is modified)
[log.txt](https://github.com/user-attachments/files/17281574/log.txt)

### To reproduce

create partitions.csv in Sketch Directory. upload big sketch with board setting default partition scheme.
https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html#using-a-custom-partition-scheme

### Expected behavior

Sketch with size < partition size should fit in app partition. Usage in % should calculated correctly by partition size from custom partitions.csv

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

10 x64 22H2

### Additional context

I have checked with ESP32 boards platform developers, they are suggesting it's either issue with Arduino IDE or usage problem.

espressif/arduino-esp32/issues/10120

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-10-07:
Hi @aIecxs. Arduino IDE doesn't know or care anything about partitions. That is a specific of the implementation of the "esp32" boards platform. Arduino IDE simply uses the abstract configuration data provided by the platform. If the platform provides inaccurate data, then that is a problem in the platform.

You can see right here where the "esp32" platform tells Arduino IDE that there is 1310720 bytes of available program memory:

https://github.com/espressif/arduino-esp32/blob/3.0.5/boards.txt#L1198

```text
esp32.upload.maximum_size=1310720
```

Note that this value is adjusted by the setting of the **Tools > Partition Scheme** menu (1310720 is the value used when you have selected **Tools > Partition Scheme > Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)**):

https://github.com/espressif/arduino-esp32/blob/3.0.5/boards.txt#L1241-L1291

```text
esp32.menu.PartitionScheme.default=Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
esp32.menu.PartitionScheme.default.build.partitions=default
esp32.menu.PartitionScheme.defaultffat=Default 4MB with ffat (1.2MB APP/1.5MB FATFS)
esp32.menu.PartitionScheme.defaultffat.build.partitions=default_ffat
esp32.menu.PartitionScheme.default_8MB=8M with spiffs (3MB APP/1.5MB SPIFFS)
esp32.menu.PartitionScheme.default_8MB.build.partitions=default_8MB
esp32.menu.PartitionScheme.default_8MB.upload.maximum_size=3342336
esp32.menu.PartitionScheme.minimal=Minimal (1.3MB APP/700KB SPIFFS)
esp32.menu.PartitionScheme.minimal.build.partitions=minimal
esp32.menu.PartitionScheme.no_fs=No FS 4MB (2MB APP x2)
esp32.menu.PartitionScheme.no_fs.build.partitions=no_fs
esp32.menu.PartitionScheme.no_fs.upload.maximum_size=2031616
esp32.menu.PartitionScheme.no_ota=No OTA (2MB APP/2MB SPIFFS)
esp32.menu.PartitionScheme.no_ota.build.partitions=no_ota
esp32.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
esp32.menu.PartitionScheme.noota_3g=No OTA (1MB APP/3MB SPIFFS)
esp32.menu.PartitionScheme.noota_3g.build.partitions=noota_3g
esp32.menu.PartitionScheme.noota_3g.upload.maximum_size=1048576
esp32.menu.PartitionScheme.noota_ffat=No OTA (2MB APP/2MB FATFS)
esp32.menu.PartitionScheme.noota_ffat.build.partitions=noota_ffat
esp32.menu.PartitionScheme.noota_ffat.upload.maximum_size=2097152
esp32.menu.PartitionScheme.noota_3gffat=No OTA (1MB APP/3MB FATFS)
esp32.menu.PartitionScheme.noota_3gffat.build.partitions=noota_3gffat
esp32.menu.PartitionScheme.noota_3gffat.upload.maximum_size=1048576
esp32.menu.PartitionScheme.huge_app=Huge APP (3MB No OTA/1MB SPIFFS)
esp32.menu.PartitionScheme.huge_app.build.partitions=huge_app
esp32.menu.PartitionScheme.huge_app.upload.maximum_size=3145728
esp32.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)
esp32.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
esp32.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080
esp32.menu.PartitionScheme.fatflash=16M Flash (2MB APP/12.5MB FATFS)
esp32.menu.PartitionScheme.fatflash.build.partitions=ffat
esp32.menu.PartitionScheme.fatflash.upload.maximum_size=2097152
esp32.menu.PartitionScheme.app3M_fat9M_16MB=16M Flash (3MB APP/9.9MB FATFS)
esp32.menu.PartitionScheme.app3M_fat9M_16MB.build.partitions=app3M_fat9M_16MB
esp32.menu.PartitionScheme.app3M_fat9M_16MB.upload.maximum_size=3145728
esp32.menu.PartitionScheme.rainmaker=RainMaker 4MB
esp32.menu.PartitionScheme.rainmaker.build.partitions=rainmaker
esp32.menu.PartitionScheme.rainmaker.upload.maximum_size=1966080
esp32.menu.PartitionScheme.rainmaker_4MB=RainMaker 4MB No OTA
esp32.menu.PartitionScheme.rainmaker_4MB.build.partitions=rainmaker_4MB_no_ota
esp32.menu.PartitionScheme.rainmaker_4MB.upload.maximum_size=4038656
esp32.menu.PartitionScheme.rainmaker_8MB=RainMaker 8MB
esp32.menu.PartitionScheme.rainmaker_8MB.build.partitions=rainmaker_8MB
esp32.menu.PartitionScheme.rainmaker_8MB.upload.maximum_size=4116480
esp32.menu.PartitionScheme.zigbee_zczr=Zigbee ZCZR 4MB with spiffs
esp32.menu.PartitionScheme.zigbee_zczr.build.partitions=zigbee_zczr
esp32.menu.PartitionScheme.zigbee_zczr.upload.maximum_size=1310720
esp32.menu.PartitionScheme.custom=Custom
esp32.menu.PartitionScheme.custom.build.partitions=
esp32.menu.PartitionScheme.custom.upload.maximum_size=16777216
```

If the custom partition provides a different amount of program memory, you would need to edit this line accordingly, and select  **Tools > Partition Scheme > Custom** from the Arduino IDE menus:

```text
esp32.menu.PartitionScheme.custom.upload.maximum_size=16777216
```

closed 2024-10-07 by per1234

aIecxs, 2024-10-07:
no, sorry. according to the documentation that's not how it works. partitioning must ensured from sketch entirely. modification of boards.txt is not acceptable because this is lost during updates.