arduino/arduino-ide #2528: [help request] how does it work the search-replace function with regex captured groups?

atesin, 2024-10-08:
### Describe the request

i need to partially replace text in many ide editor tabs, i am trying to use the search-replace function (the sidebar magnifying glass) with regex captured groups, like this example for identify print commands just to add `\r\n` at the end of printed sting

search regex: `print\("(.*)"\);`
replace regex: `print("\1\\r\\n");`

i captured the content with `(.*)` then i tried to include it in replacement with `\1` and `$1` but it prints them literally

p.s.: i haven't found any info on this anywhere... please also make a help page somewhere

### Describe the current behavior

i tried to use captured content with `\1` and `$1` but it prints them literally

### Arduino IDE version

2.2.3

### Operating system

Windows

### Operating system version

win 10 pro x64 22h2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-10-09:
Hi @atesin. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2024-10-09 by per1234

atesin, 2024-10-09:
yes but ...  i found NO INFO ON THE FORUM (as you suggested), or in the issue tracker, or in docs or anywhere!

you make us to tick some checkboxes after post an issue checking other issues, the forum, the docs, etc, before post....  and i had search all day, found nothing, mark the checkboxes, and dare to post the issue

if dont match the rules exactly, consider this issue as a request to post this info somewere easily accesible in documentation(s)

and as we are talking about this, could you also be kind and tell me how to use regex capture groups in search-replace function in ide?? .. thanks

per1234, 2024-10-09:
@atesin you have to make a post on the forum to request assistance.