arduino/arduino-ide #2536: IDE does not start - no window

sclsj, 2024-10-22:
### Describe the problem

IDE starts, but does not present the main window

```
jin@Joy-MBP ~> "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
Arduino IDE 2.3.3
Checking for frontend application configuration customizations. Module path: /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/electron-main.js, destination 'package.json': /Applications/Arduino IDE.app/Contents/Resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.3" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.0.4" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-09-25T09:35:36.798Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.3","cliVersion":"1.0.4","buildDate":"2024-09-25T09:35:36.798Z"}
Starting backend process. PID: 5292
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///Users/jin/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2024-10-22T01:52:38.564Z root INFO Backend Object.initialize: 1.9 ms [Finished 0.273 s after backend start]
2024-10-22T01:52:38.564Z root INFO Backend Object.configure: 1.4 ms [Finished 0.273 s after backend start]
2024-10-22T01:52:38.564Z root INFO Backend a.onStart: 0.7 ms [Finished 0.273 s after backend start]
2024-10-22T01:52:38.565Z root INFO Backend a.configure: 0.1 ms [Finished 0.286 s after backend start]
2024-10-22T01:52:38.565Z root INFO Backend d.onStart: 0.1 ms [Finished 0.286 s after backend start]
2024-10-22T01:52:38.566Z root INFO Backend d.configure: 0.1 ms [Finished 0.286 s after backend start]
2024-10-22T01:52:38.566Z root INFO Backend d.initialize: 13.0 ms [Finished 0.286 s after backend start]
2024-10-22T01:52:38.566Z root INFO Backend c.onStart: 0.1 ms [Finished 0.286 s after backend start]
2024-10-22T01:52:38.566Z root INFO Backend s.configure: 0.5 ms [Finished 0.287 s after backend start]
2024-10-22T01:52:38.567Z config INFO >>> Initializing CLI configuration...
2024-10-22T01:52:38.567Z root INFO Backend c.initialize: 0.2 ms [Finished 0.287 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend x.onStart: 0.4 ms [Finished 0.287 s after backend start]
2024-10-22T01:52:38.567Z config INFO Loading CLI configuration from /Users/jin/.arduinoIDE/arduino-cli.yaml...
2024-10-22T01:52:38.567Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli...
2024-10-22T01:52:38.567Z root INFO Backend s.initialize: 0.2 ms [Finished 0.288 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend w.onStart: 0.5 ms [Finished 0.288 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend r.configure: 0.9 ms [Finished 0.288 s after backend start]
2024-10-22T01:52:38.567Z discovery-log INFO start
2024-10-22T01:52:38.567Z discovery-log INFO start new deferred
2024-10-22T01:52:38.567Z root INFO Backend d.initialize: 0.2 ms [Finished 0.288 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend y.onStart: 0.8 ms [Finished 0.289 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend r.configure: 0.7 ms [Finished 0.289 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend s.onStart: 0.2 ms [Finished 0.289 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend h.configure: 0.2 ms [Finished 0.289 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend a.configure: 0.0 ms [Finished 0.289 s after backend start]
2024-10-22T01:52:38.567Z root INFO Backend l.configure: 0.0 ms [Finished 0.290 s after backend start]
2024-10-22T01:52:38.569Z root ERROR Failed to start the backend application:
2024-10-22T01:52:38.569Z root ERROR Error: listen EADDRNOTAVAIL: address not available 10.53.146.123
    at Server.setupListenHandle [as _listen2] (node:net:1734:21)
    at listenInCluster (node:net:1799:12)
    at GetAddrInfoReqWrap.doListen [as callback] (node:net:1948:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:110:8)
2024-10-22T01:52:38.569Z root ERROR Uncaught Exception:  Error: listen EADDRNOTAVAIL: address not available 10.53.146.123
2024-10-22T01:52:38.569Z root ERROR Error: listen EADDRNOTAVAIL: address not available 10.53.146.123
    at Server.setupListenHandle [as _listen2] (node:net:1734:21)
    at listenInCluster (node:net:1799:12)
    at GetAddrInfoReqWrap.doListen [as callback] (node:net:1948:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:110:8)
2024-10-22T01:52:38.571Z root WARN The local plugin referenced by local-dir:/Users/jin/.arduinoIDE/plugins does not exist.
2024-10-22T01:52:38.571Z root WARN The local plugin referenced by local-dir:/Users/jin/.arduinoIDE/extensions does not exist.

^CStored workspaces roots: 
No sketches were scheduled for deletion.
Could not terminate the backend process. It was not running.
```

### To reproduce

Open the application

### Expected behavior

Main window opens

[... 166 lines cut ...]
	inet6 fdb2:2c26:f4e4:1::1 prefixlen 64 
	Configuration:
		id 0:0:0:0:0:0 priority 0 hellotime 0 fwddelay 0
		maxage 0 holdcnt 0 proto stp maxaddr 100 timeout 1200
		root id 0:0:0:0:0:0 priority 0 ifcost 0 port 0
		ipfilter disabled flags 0x0
	member: vmenet1 flags=3<LEARNING,DISCOVER>
	        ifmaxaddr 0 port 28 priority 0 path cost 0
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
en10: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether d2:d1:71:23:09:95 
	inet6 fe80::d0d1:71ff:fe23:995%en10 prefixlen 64 scopeid 0x7 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
```

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-10-22:
Hi @sclsj. This is caused by a problem with your system, rather than Arduino IDE. I think it will be more appropriate for us to assist you in solving the problem over on Arduino Forum:

https://forum.arduino.cc/c/software/ide-2-x/93

Make a forum topic and we'll help you out over there.

sclsj, 2024-10-22:
I'm having some trouble signing up for the Arduino Forum. Can you elaborate why you determined that this is caused by a problem with my system?

sclsj, 2024-10-22:
Just let it sit there for 10 minutes and it suddenly went through. 

https://forum.arduino.cc/t/ide-does-not-start-eaddrnotavail-address-not-available/1313854

closed 2024-10-22 by per1234