arduino/arduino-ide #2541: Arduino Due on Macbook with M1 install the wrong toolchain

antoniocaruso, 2024-10-25:
### Describe the problem

Seems the binary for g++ is wrong.

Errore:

FQBN: arduino:sam:arduino_due_x
Utilizzo della scheda 'arduino_due_x' dalla piattaforma nella cartella: /Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Utilizzo del core 'arduino' dalla piattaforma nella cartella: /Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12

Rilevamento delle librerie utilizzate in corso...
/Users/antonio/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/antonio/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /private/var/folders/kq/cm7mwl0n6ds1_t_q668r_jbr0000gn/T/arduino/sketches/EB39F2F284879E19197DA52AA0DAC5D8/sketch/iotschednew.ino.cpp -o /dev/null
fork/exec /Users/antonio/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: bad CPU type in executable

Compilation error: fork/exec /Users/antonio/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: bad CPU type in executable

### To reproduce

Clean Macbook.
Install Arduino IDE 2.3.3
Attach Arduino Due to USB.
Board is recognized, load Blink example.
The IDE ask correctly to install the toolchain for the board, arduino:sam
ToolChain Installed.
Compile Blink.

### Expected behavior

Working.

### Arduino IDE version

2.3.3

### Operating system

macOS

### Operating system version

Sequoia 15.0.1

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-10-26:
Hi @antoniocaruso. The "arm-none-eabi-gcc" compiler toolchain used to compile sketches for the Arduino Due board is only distributed in builds for "Intel" (x86) CPUs. This is not a significant problem for those of us with macOS machines that have Apple Silicon CPUs because Apple provides the [**Rosetta 2** binary translator](https://support.apple.com/HT211861) that allows binaries built for x86 host architectures to run on Apple Silicon CPUs.

The reason why you encountered this error is because you don't have **Rosetta 2** installed. macOS is smart enough to notify you that you need to install **Rosetta 2** when you start an application that requires it. Unfortunately it is not smart enough to do that when Arduino IDE invokes an x86 executable (as it does when you compile for the Arduino Due).

So the solution will be for you to install **Rosetta 2**. You can do that by following the instructions here:

https://support.arduino.cc/hc/en-us/articles/7765785712156-Error-bad-CPU-type-in-executable-on-macOS#manual-installation

closed 2024-10-26 by per1234

antoniocaruso, 2024-10-26:
Si risolto ma sarebbe utile un toolchain nativo.

-----------------
     Prof. Antonio Caruso
     Dipartimento di Matematica e Fisica 'Ennio De Giorgi', campus Ecotekne
     Palazzo Fiorini, Università del Salento, 73100 Lecce, Italia.
     http://bilbo.unisalento.it/antonio
-----------------