arduino/arduino-ide #2544: IDE fails to start - JSON parse error

ondra-novak, 2024-10-28:
Often i am unable to start  IDE with following error
It is random behaviour. 

```
2024-10-28T16:06:29.021Z config ERROR Failed to initialize the CLI configuration. SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at x.getFallbackCliConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:763042)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async x.loadCliConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:762457)
    at async x.initConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:760906)
```

Daemon path
```
2024-10-28T16:06:28.996Z daemon INFO Starting daemon from /home/ondra/arduino-ide/resources/app/lib/backend/resources/arduino-cli...
```

Spawn command OK result:

```
ondra@nikola:~/arduino-ide$ ./resources/app/lib/backend/resources/arduino-cli config get directories --json
{
  "data": "/home/ondra/.arduino15",
  "downloads": "",
  "user": "/home/ondra/Arduino"
}

````

(error line : https://github.com/arduino/arduino-ide/blob/3ccc864453866c9273428779680e4fe2b91b1ce6/arduino-ide-extension/src/node/config-service-impl.ts#L228)

ondra-novak, 2024-10-28:
Additional:

I was able to redirect arduino-cli command line and output to a log file, verified that ide is sometime able to start. Following logs record from failed start. Nothing abnormal, still getting JSON parse error (probably a race condition)

snippet of arduino-ide log  (time in UTC)
```
024-10-28T16:54:59.920Z config INFO Loaded CLI configuration: {"board_manager":{"additional_urls":[]}}
2024-10-28T16:54:59.920Z config INFO Loading fallback CLI configuration to get 'directories.data' and 'directories.user'
2024-10-28T16:54:59.929Z config ERROR Failed to initialize the CLI configuration. SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at x.getFallbackCliConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:763042)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async x.loadCliConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:762457)
    at async x.initConfig (/home/ondra/arduino-ide/resources/app/lib/backend/main.js:2:760906)
2024-10-28T16:54:59.940Z daemon INFO time="2024-10-28T17:54:59+01:00" level=info msg="arduino-cli version 1.0.4"
time="2024-10-28T17:54:59+01:00" level=info msg="Using config file: /home/ondra/.arduinoIDE/arduino-cli.yaml"
time="2024-10-28T17:54:59+01:00" level=info msg="Executing `arduino-cli daemon`"
Daemon is now listening on 127.0.0.1:41445
{"IP":"127.0.0.1","Port":"41445"}
2024-10-28T16:54:59.940Z daemon INFO Daemon is running.
2024-10-28T16:54:59.962Z root INFO Resolve plugins list: 61.8 ms [Finished 0.645 s after backend start]
```
Captured log redirected from arduino-cli (time in CET - +01:00)
```
Starting /home/ondra/arduino-ide/resources/app/lib/backend/resources/arduino-cli-2 config dump --json
Starting /home/ondra/arduino-ide/resources/app/lib/backend/resources/arduino-cli-2 config get directories --json
{
  "config": {}
}
Exited: 0
Starting /home/ondra/arduino-ide/resources/app/lib/backend/resources/arduino-cli-2 daemon --port 0 --config-file /home/ondra/.arduinoIDE/arduino-cli.yaml -v
{
  "data": "/home/ondra/.arduino15",
  "downloads": "",
  "user": "/home/ondra/Arduino"
}
Exited: 0
time="2024-10-28T17:54:59+01:00" level=info msg="arduino-cli version 1.0.4"
time="2024-10-28T17:54:59+01:00" level=info msg="Using config file: /home/ondra/.arduinoIDE/arduino-cli.yaml"
time="2024-10-28T17:54:59+01:00" level=info msg="Executing `arduino-cli daemon`"
Daemon is now listening on 127.0.0.1:41445
{"IP":"127.0.0.1","Port":"41445"}

time="2024-10-28T17:54:59+01:00" level=info msg="Checking signature" index=/home/ondra/.arduino15/package_index.json signatureFile=/home/ondra/.arduino15/package_index.json.sig trusted=true
time="2024-10-28T17:54:59+01:00" level=info msg="Loading hardware from: /home/ondra/.arduino15/packages"
```

Redirection script:

```
#!/bin/bash
ABSDIR=`dirname "$0"`
echo "Starting $ABSDIR/arduino-cli-2 $*" >> /tmp/arduino-cli.log
$ABSDIR/arduino-cli-2 $* | tee -a /tmp/arduino-cli.log
echo "Exited: $?" >> /tmp/arduino-cli.log
```

Twilight-Logic, 2024-10-29:
I have the same random issue. Its seems also to get stuck randomly when installing updated board and other libraries.
This is par of my log:

```
Warning: loader_get_json: Failed to open JSON file intel_hasvk_icd.x86_64.json
Warning: loader_get_json: Failed to open JSON file virtio_icd.x86_64.json
Warning: /usr/lib/x86_64-linux-gnu/libvulkan_intel.so: cannot open shared object file: Permission denied
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/x86_64-linux-gnu/libvulkan_intel.so. Ignoring this JSON
Warning: loader_get_json: Failed to open JSON file lvp_icd.x86_64.json
Warning: /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so: cannot open shared object file: Permission denied
Warning: loader_icd_scan: Failed loading library associated with ICD JSON /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so. Ignoring this JSON
Warning: loader_scanned_icd_add: Could not get 'vkCreateInstance' via 'vk_icdGetInstanceProcAddr' for ICD libGLX_nvidia.so.0
Warning: vkCreateInstance: Found no drivers!
Warning: vkCreateInstance failed with VK_ERROR_INCOMPATIBLE_DRIVER
    at CheckVkSuccessImpl (../../third_party/dawn/src/dawn/native/vulkan/VulkanError.cpp:88)
    at CreateVkInstance (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:458)
    at Initialize (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:344)
    at Create (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:266)
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:521)
```

Mentek44, 2025-02-09:
Hi, same issue here on EndeavourOS with kernel 6.12.12-1-lts. Removing/renaming the `~/.arduinoIDE/arduino-cli.yaml` file solved it for me.

```
[16:41:56] [mentek44@telephasic-workshop:~] 
0 $ cat .arduinoIDE/arduino-cli.yaml 
board_manager:
    additional_urls: []
```

```
[16:42:29] [mentek44@telephasic-workshop:~] 
0 $ mv .arduinoIDE/arduino-cli.yaml .arduinoIDE/arduino-cli.yaml.bak
```
after starting `arduino-ide` it recreated the config:
```
[16:44:19] [mentek44@telephasic-workshop:~] 
0 $ cat .arduinoIDE/arduino-cli.yaml
board_manager:
    additional_urls: []
```
`diff` tool does not detect any differences tho, very strange:
```
[16:44:23] [mentek44@telephasic-workshop:~] 
0 $ diff .arduinoIDE/arduino-cli.yaml .arduinoIDE/arduino-cli.yaml.bak
```