arduino/arduino-ide #2551: Canno choose another board

deralteherr, 2024-10-31:
### Describe the problem

If I want to select another board with the IDE 2.3.3, the window gets completely empty except for the ‘Cancel’ and ‘OK’ buttons. Nothing works any more. This happens as soon as I enter a letter in the search field. Selecting a different port works without any problems.
My system: Intel Apple iMac, 27’, System: Ventura 13.7.1.

[image]
[image]

### To reproduce

Choose another board and port. 

### Expected behavior

I van see the boards to choose

### Arduino IDE version

2.3.3

### Operating system

macOS

### Operating system version

13.7.1

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-10-31:
Hi @deralteherr. It might be caused by https://github.com/arduino/arduino-cli/issues/2719.

Please try this:

1. Start Arduino IDE if it is not already running.
1. Check to see if there is a "**Serial Monitor**" tab in the bottom panel of the Arduino IDE window. If there is, click the **X** icon on the tab to close it:
   [image]
1. Try using the "**Anderes Board und anderen Ports wählen**" (AKA "**Select Other Board and Port**") dialog again, just as you did before when you encountered the fault.
1. Add a comment here on the issue thread to let us know whether the fault still occurs when **Serial Monitor** is closed.

deralteherr, 2024-11-01:
The error persists. Even if the serial monitor is closed.

tatertacoma, 2025-01-20:
I am also experiencing this issue on v2.3.4
I think it has to do with certain boards.

I have the CubeCell Development Framework by Heltec installed via the Board Manager
If I open the Selection dialogue no boards appear and if I search for boards that don't (seem to) include any Heltec boards, it acts normally, but if I type anything(letter) in that would result in a "bugged" board, it breaks as shown above.
And if I remove the Heltec Boards using the board manager, it also acts just fine with anything I type into it.

I found the work-around to just use Tools>Board and selecting it that way.

vkaspar, 2025-04-25:
e problem with Arduino IDE 2.3.6 and CubeCell 1.5