arduino/arduino-ide #2558: Serial Monitor text selection problem

ivannunyadambiz, 2024-11-11:
### Describe the problem

When selecting text in serial monitor, the selection escalates beyond the Serial monitor into the code line numbers.
Also, when selecting some text in the Serial monitor and pressing [CTRL] + [A], the code gets selected, not the text in the Serial monitor.

### To reproduce

see problem description

### Expected behavior

well... not that.

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

11

### Additional context

[image]

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-11-11:
Hi @ivannunyadambiz . Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/812.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2024-11-11 by per1234

ivannunyadambiz, 2024-11-11:
wait, what? it's "not planned" to fix the bug of selecting the line numbers when text in the monitor is selected?
you're kidding, right?