arduino/arduino-ide #2561: Crashing on start

Aphexus, 2024-11-13:
### Describe the problem

Failed to start the electron application.
Error: EISDIR: illegal operation on a directory, realpath 'Z:\Temp'
    at Function.<anonymous> (node:fs:2719:3)
    at Function.native (node:electron/js2c/asar_bundle:2:5101)
    at new d (C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:2:769052)
    at Ee (C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:633790)
    at C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:637767
    at Le (C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:637836)
    at C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:638223
    at Fe (C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:638164)
    at Oe (C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:638198)
    at C:\Apps\Programming\ArduinoIDE\resources\app\lib\backend\electron-main.js:8:636512 {
  errno: -4068,
  syscall: 'realpath',
  code: 'EISDIR',
  path: 'Z:\\Temp'
}

The Z drive I use is actually a ram drive  which I created using ImDisk. The reason is, of course, even though I have 128GB of memory apps love to thrash the drives temp files so I use ramdisk to avoid this and it typically speeds up some applications.

### To reproduce

Do as you like

### Expected behavior

For it to work.

### Arduino IDE version

nightly

### Operating system

Windows

### Operating system version

Latest

### Additional context

This technically may be an issue with ImDisk or with node or whatever Arduino uses.

E.g., every once in a while I have an issue with some apps not working well with the temp dir being on a ramdisk. E.g., some recycle programs do not work well because the drive is not NTFS(does not contain a $recycle dir).

There technically should be no reason it shouldn't work though. I use thousands of programs and almost all work fine(from more complex apps like various IDE's to graphics apps to whatever.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

Aphexus, 2024-11-13:
I was able to "solve" this issue by temporarily changing the TEMP env variable to a different drive.

OzFalcon, 2024-11-14:
Have you tried formatting the Ram disk as NTFS?

> imdisk -a -s 512M -m X: -p "/fs:ntfs /q /y"

    -a initializes the virtual disk.
    -s 512M is the size, 512 MegaBytes.
    The full choices are b, k, m, g, t, K, M, G, or T.
    These denote a number of 512-byte blocks, thousand bytes, million bytes, billion bytes, trillion bytes, KB, MB, GB, and TB, respectively.
    -m X: sets up the mount point a.k.a. the drive letter, X:.
    -p "fs:ntfs /q /y" formats the drive.
    -p's parameters are actually for Windows' format program.
    So, if you want the RAM disk in a different filesystem, just change ntfs to fat (FAT16) or fat32 (FAT32).

Ref: https://forums.guru3d.com/threads/guide-using-imdisk-to-set-up-ram-disk-s-in-windows-with-no-limit-on-disk-size.356046/

Note: I use Linux so the above was just a google search for "imdisk ntfs"

BariSaxGreg, 2025-01-30:
Issue can be solved / worked around by copying all files from the original windows temp directory into the initialized RAM disk temp folder. I'm no programming expert but it would appear that the IDE holds important program files in the temp directory. I had previously copied over the arduino language server folders and that didn't fix it from my memory, but something in there holds the key to this issue.

closed 2025-10-02 by per1234