arduino/arduino-ide #2575: IDE not compiling if Board manager URL is 404

CWCarlson123, 2024-11-22:
### Describe the problem

If the URL (or any path of the URL) in preferences menu for additional boards returns a 404 causes the IDE to not compile. This is indicated by a notification of downloading the package that does not progress.

### To reproduce

1. open a sketch.
2. Enter a broken URL into the additional board managers URL in the preferences: https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json (was broken when I tried this).
3. attempts to download package.
4. try to compile

### Expected behavior

IDE will not compile, it will just sit there with the notification that it is compiling, but never completes.

### Arduino IDE version

2.3.3

### Operating system

Windows

### Operating system version

windows 11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

metehoca, 2024-11-28:
i have exactly same problem on win10

per1234, 2024-11-30:
Thanks for your report @CWCarlson123. I am not able to reproduce the fault.

However, your description:

> a notification of downloading the package that does not progress.

> IDE will not compile, it will just sit there with the notification that it is compiling, but never completes.

matches exactly with a known bug in Arduino IDE 2.3.3: https://github.com/arduino/arduino-ide/issues/2562. This is caused by having Serial Monitor or Serial Plotter open while the IDE performs an index downloaded. It occurs regardless of the success or failure of the index download.

https://github.com/arduino/arduino-ide/issues/2562 has already been fixed. The fix was made after the time of the 2.3.3 release, so it is currently only available when using the latest nightly build of Arduino IDE.

@CWCarlson123 and @metehoca please try again using the nightly build of Arduino IDE and then add a comment here to tell us whether the problem of the hanging compilation still occurs. You can download the nightly build from the links listed here:

https://www.arduino.cc/en/software#nightly-builds

You can have Arduino IDE 2.3.3 and the nightly build installed on your computer at the same time without any problems, so you don't need to uninstall Arduino IDE 2.3.3 before installing the nightly.

metehoca, 2024-11-30:
#2562  seems to match this description, but it’s incomplete. The issue persists even without the Serial Monitor being open and also occurs immediately when the IDE is launched. During this time, the "Downloading" balloon gets stuck, and the IDE cannot compile.

As a temporary workaround, I open a new sketch and close the old window. From what I understand, the issue seems to freeze after the last link in the board list is processed.

per1234, 2024-11-30:
Thanks for the information @metehoca. Please try again using the nightly build of Arduino IDE and then add a comment here to tell us whether the problem of the hanging compilation still occurs. You can download the nightly build from the links listed here:

https://www.arduino.cc/en/software#nightly-builds

You can have Arduino IDE 2.3.3 and the nightly build installed on your computer at the same time without any problems, so you don't need to uninstall Arduino IDE 2.3.3 before installing the nightly.

CWCarlson123, 2024-12-02:
Thanks for clarification on the issue,

I have a better observation of the issue, based on @metehoca's comments:

1. opened digtialRead serial example. Change input pin type to INPUT_PULLUP;
2. compile and upload sketch to Nanoevery;
3. Open serial monitor;
4. open serial plotter;
5. With the **serial plotter** window open, roll the Arduino AVR boards version back to 1.8.5 from the board manager, the download indication balloon just stays there.
6. The serial plotter window is closed. 
7. When I attempt the re-compile the code, the "Compiling sketch..." balloon appears and compiling does not complete.
8. Close the **serial monitor** tab above the output window - **the download completes and then the the project then compiles.**

per1234, 2024-12-04:
Thanks for the update. Based on the new information, I see this is a duplicate of https://github.com/arduino/arduino-ide/issues/2562 / https://github.com/arduino/arduino-cli/issues/2719. The bug was fixed by https://github.com/arduino/arduino-ide/pull/2571, which was released in Arduino IDE 2.3.4. So just update your IDE to 2.3.4 and you will be set.

closed 2024-12-04 by per1234