arduino/arduino-ide #2582: color impossible to change

RayofJay, 2024-11-26:
### Describe the request

I'm just flabbergasted...how is it some god damn hard to change the IDE colors?  visual studio .net SO EASY, albeit a lot of items to sift through, but EASY.  aruidno IDE?  non-eixstent.  I really can't convey my disappointment and anger here.  I've wasted over an hour and have gotten nowhere.  Complete fail.

### Describe the current behavior

colors in the IDE are impossible to change

### Arduino IDE version

2.4.3

### Operating system

Windows

### Operating system version

win 11

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

per1234, 2024-11-27:
Hi @RayofJay. We provide multiple themes which can be selected from the Arduino IDE preferences. In addition, you can install any of the thousands of 3rd party free open source pre-made VS Code theme extensions, as documented here:

https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#3rd-party-themes

Note that, although Arduino does officially support the installation of theme extensions in general, we can not guarantee the functionality of 3rd party extensions as we do not have any involvement in the development of those projects.

This issue tracker is only to be used to report bugs or feature requests specific to the project. This is not an appropriate place to request assistance with customizing the colors of Arduino IDE. The appropriate place to request such assistance is on Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/software/ide-2-x/93

closed 2024-11-27 by per1234

RayofJay, 2024-11-28:
This is not a bug report or request for assistance.  it is a direct and harsh slam against the designers of the Arduino interface.  I and still just flabbergasted that I cannot change the colors whenever i wish....

https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#3rd-party-themes

https://forum.arduino.cc/c/software/ide-2-x/93