arduino/arduino-ide #2585: Arduino IDE on First Run is a Stupid Interface

ruvsudrn, 2024-12-02:
### Describe the problem

When running the Arduino IDE for the first time it shows a white window with a "pulsing" Arduino logo in the middle.

From reading other complaints and comments online the program is supposed to be downloading library files and setting up for first run.

The interface does not communicate any of this to the user and as such the interface is "stupid by design" and should be improved.

The addition of a simple text box at the bottom of the window with some status information would help every single user to be patient and wait for the IDE to complete its tasks.

[image]

### To reproduce

If needed uninstall Arduino and purge all old Arduino files.

Run the Arduino IDE for the first time.

### Expected behavior

The program should start.

If the program needs to do stuff, then it should tell the user what it is doing.

This is interface design 101 guys, c'mon.

### Arduino IDE version

Any

### Operating system

N/A

### Operating system version

Any

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

ruvsudrn, 2024-12-03:
I have also noted that this interface is also used when performing a long running task, such as opening or closing a file/sketch. This also fails basic interface design guidelines as there is no progress bar or information tracker to assist the user to understand what the software is doing.

Dmitry-Chernikov, 2024-12-11:
I launch it via CLI ./arduino-ide and it launches fine, but when I launch it from the desktop shortcut, the same program login window just hangs there.https://github.com/user-attachments/assets/cebe4d6e-e886-435e-bea4-c360fcadc23e
```shell
dmitry@PC3426:~/arduino-ide$ ./arduino-ide 
Arduino IDE 2.3.4
Checking for frontend application configuration customizations. Module path: /home/dmitry/arduino-ide/resources/app/lib/backend/electron-main.js, destination 'package.json': /home/dmitry/arduino-ide/resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.4" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.1.1" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-12-03T10:51:12.539Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.4","cliVersion":"1.1.1","buildDate":"2024-12-03T10:51:12.539Z"}
MESA-INTEL: warning: Haswell Vulkan support is incomplete
Starting backend process. PID: 16958
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/dmitry/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2024-12-11T07:36:45.757Z root INFO Backend Object.initialize: 6.2 ms [Finished 0.632 s after backend start]
2024-12-11T07:36:45.757Z root INFO Backend Object.configure: 5.3 ms [Finished 0.632 s after backend start]
2024-12-11T07:36:45.757Z root INFO Backend s.onStart: 3.7 ms [Finished 0.632 s after backend start]
2024-12-11T07:36:45.758Z root INFO Backend s.configure: 0.4 ms [Finished 0.660 s after backend start]
2024-12-11T07:36:45.759Z root INFO Backend d.onStart: 0.3 ms [Finished 0.660 s after backend start]
2024-12-11T07:36:45.760Z root INFO Backend d.configure: 0.1 ms [Finished 0.660 s after backend start]
2024-12-11T07:36:45.761Z root INFO Backend d.initialize: 28.3 ms [Finished 0.661 s after backend start]
2024-12-11T07:36:45.761Z root INFO Backend c.onStart: 0.3 ms [Finished 0.661 s after backend start]
2024-12-11T07:36:45.761Z root INFO Backend a.configure: 0.6 ms [Finished 0.661 s after backend start]
2024-12-11T07:36:45.763Z config INFO >>> Initializing CLI configuration...
2024-12-11T07:36:45.763Z root INFO Backend c.initialize: 0.5 ms [Finished 0.662 s after backend start]
2024-12-11T07:36:45.763Z root INFO Backend x.onStart: 1.1 ms [Finished 0.663 s after backend start]
2024-12-11T07:36:45.764Z config INFO Loading CLI configuration from /home/dmitry/.arduinoIDE/arduino-cli.yaml...
2024-12-11T07:36:45.765Z daemon INFO Starting daemon from /home/dmitry/arduino-ide/resources/app/lib/backend/resources/arduino-cli...
2024-12-11T07:36:45.765Z root INFO Backend a.initialize: 0.6 ms [Finished 0.664 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend w.onStart: 1.2 ms [Finished 0.665 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend r.configure: 2.0 ms [Finished 0.665 s after backend start]
2024-12-11T07:36:45.765Z discovery-log INFO start
2024-12-11T07:36:45.765Z discovery-log INFO start new deferred
2024-12-11T07:36:45.765Z root INFO Backend d.initialize: 0.5 ms [Finished 0.665 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend y.onStart: 2.3 ms [Finished 0.667 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend r.configure: 2.0 ms [Finished 0.667 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend a.onStart: 0.5 ms [Finished 0.669 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend h.configure: 0.4 ms [Finished 0.669 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend s.configure: 0.1 ms [Finished 0.669 s after backend start]
2024-12-11T07:36:45.765Z root INFO Backend l.configure: 0.1 ms [Finished 0.669 s after backend start]
2024-12-11T07:36:45.768Z root INFO Theia app listening on http://127.0.0.1:38019.
2024-12-11T07:36:45.768Z root INFO Finished starting backend application: 2.7 ms [Finished 0.672 s after backend start]
2024-12-11T07:36:45.769Z root WARN The local plugin referenced by local-dir:/home/dmitry/.arduinoIDE/plugins does not exist.
2024-12-11T07:36:45.769Z root WARN The local plugin referenced by local-dir:/home/dmitry/.arduinoIDE/extensions does not exist.
2024-12-11T07:36:45.780Z config INFO Loaded CLI configuration: {"board_manager":{"additional_urls":[]},"build_cache":{"compilations_before_purge":10,"ttl":"720h0m0s"},"daemon":{"port":"50051"},"directories":{"builtin":{"libraries":"/home/dmitry/.arduino15/libraries"},"data":"/home/dmitry/.arduino15","downloads":"/home/dmitry/.arduino15/staging","user":"/home/dmitry/ProjectsArduino/"},"library":{"enable_unsafe_install":false},"locale":"r [... cut ...]
Marking workspace as a closed sketch. Workspace URI: file:///home/dmitry/ProjectsArduino/3g833_nov7a. Date: 1733902723919.
Closing channel on service path '/services/electron-window'.
Closing channel on service path '/services/ide-updater'.
Storing the sketch as a workspace root: </home/dmitry/ProjectsArduino/3g833_nov7a>.
Skipped storing sketch as workspace root. Already visited: </home/dmitry/ProjectsArduino/3g833_nov7a>.
Stored workspaces roots: /home/dmitry/ProjectsArduino/3g833_nov7a
No sketches were scheduled for deletion.

```