arduino/arduino-ide #2592: Bluescreen of Death with Serial Monitor

Joko-1, 2024-12-05:
### Describe the problem

**Describe the problem**

Bluescreen of Death after about 30sec in Serial Monitor with continuous serial prints (about every 10ms)
Stillstandcode: DRIVER IRQL NOT. LESS OR EQUAL
Fehlerursache: CH341S64.SYS

[image]

When I open the serial monitor over Putty, I have no problems

**Solution attempts to fix the problem.**s
Re Installing the CH340 driver,
Updating from 2.3.3 to 2.3.4,
Using another cable,
Using another Arduino script.

**Board**: ESP32-WROOM-32 with **CH340 Serial chip** 

### To reproduce

Serial Monitor of ESP 32 with CH340 chip with continuous serial prints (about every 10ms)

### Expected behavior

Normal Serial Monitor

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

Windown 10 22H2

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-12-05:
Hi @Joko-1. Thanks for your report. I'm going to ask you to provide some additional information:

Please provide a simple sketch that you have verified to produce the BSOD. For example, does the BSOD occur if you upload this sketch to your board?:

```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("hello");
  delay(5);
}
```

---

Please also tell me which version of the CH340 driver you have installed. You can find this information by the following procedure:

1. Open the [Windows **Device Manager**](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Select **View > Devices by type** from the **Device Manager** menus.
1. Open the "**View**" menu.
1. If there is a **✓** to the left of the "**Show hidden devices**" menu item, click on "**Show hidden devices**" to disable it.
1. Open the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. You should see a port identified as "**USB-SERIAL CH340 (COM<i>n</i>)**" under the "**Ports (COM & LPT)**" section (where "COM<i>n</i>" is some serial port e.g., COM4). Double click on that item.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Driver**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.

You will see the version in the "**Driver Version**" field of the dialog.

---

Please also tell me whether there is labeling on the CH340 chip on your board. If present, the labeling would look something like this:

[image]

<sub>

<details>
  <summary>📷</summary>
  <p>

[SparkFun](https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all) - [CC BY-SA 4.0](https://creativecommons.org/licenses/by-sa/4.0/) (cropped)

  </p>
</details>

</sub>

But the chip might also be blank.

Joko-1, 2024-12-05:
Thanks for respond,

**It also happens, when I use your script.**

I tested it a few times with baud rate 115200, 9600.
I didn't get the 30s as earlier (it took ~2-3min), but I noticed that it happened faster at 11520 baud.

**Driver:**
Driver Provider: wch.cn
Driver Date: 16.09.2024
Driver Version: 3.9.2024.9

**Chip:**
WCH
CH340C
205695D41

KashifKhaan, 2024-12-06:
The BSOD issue appears to stem from the **CH341S64.SYS** driver for the CH340 Serial chip. Here's a streamlined approach to resolve it:

- **Update or Revert the CH340 Driver:** Download the latest or a stable older version from wch.cn.
- **Adjust Serial Communication:** Reduce the baud rate (e.g., to 9600) and limit the frequency of serial prints (e.g., every 100ms or more).
- **Use Alternative Serial Tools:** If the Arduino IDE causes BSOD, switch to PuTTY or another terminal.
- **Test Different Hardware:** Use a different USB cable, port, or a board with a non-CH340 USB-to-Serial chip (e.g., CP2102).
- **Update Arduino IDE:** Ensure you're on the latest version to avoid IDE-specific bugs.

AkhmadKhoirudin, 2024-12-06:
same with me, tooSame with me, I even almost reinstalled Arduino Ide several times

bankdaysun, 2024-12-09:
After this issue occurred, I tried to fix it by installing the old 2011 driver
[image]
Select "Older Windows Driver Version and Instructions"
https://sparks.gogo.co.nz/ch340.html?srsltid=AfmBOorOaQq3i3DrMbWlf7fW4zSXS1gbigIbjYHjBQ3w1P91_VicE93r
It worked fine and stopped the BSOD
Device: ESP32 Dev module CH340
date:9/12/2024

per1234, 2025-01-06:
Thanks for sharing this valuable information @bankdaysun! This indicates that the problem is in the latest version of the CH340 driver, and not in Arduino IDE.

Did you find that the problem came back later? I ask because in my experience Windows will automatically update the driver to the problematic version once again after some time.

per1234, 2025-01-07:
Hi @Joko-1. I will provide instructions you can follow to install an older version of the driver. Hopefully the problem will not occur when that older version is used, as was reported by **@bankdaysun**.

Unfortunately I am not able to reproduce the fault on a Windows 10 virtual machine, or on a Windows 11 physical machine (I don't have a Windows 10 physical machine) by running the "hello" sketch on my ESP32-CAM board (which has a CH340 chip) with the problematic `3.9.2024.9` version of the driver installed. So I am not able to personally verify that this workaround will fix the problem.

1. Click the following link to download the previous version of the driver from the chip manufacturer's website:
   <https://www.wch-ic.com/downloads/file/65.html?time=2023-03-16%2022:57:59>
1. Wait for the download to finish.
1. Run the downloaded `CH341SER.EXE` file.
1. A "**User Account Control**" dialog may now appear asking "**Do you want to allow this app to make changes to your device?**". Click the "**Yes**" button.
1. A "**DriverSetup(X64)**" dialog will open. Click the "**INSTALL**" button in the dialog.
1. Wait for the driver installation to finish, as indicated by the appearance of a "**Driver install success!**" dialog.
1. Click the "**OK**" button in the "**Driver install success!**" dialog.
   The dialog will close.
1. Click the **X** icon at the top right of the "**DriverSetup(X64)**" dialog
   The dialog will close.
1. Close Arduino IDE if it is running.
1. Connect the Arduino board to your computer with a USB cable.
1. Open the [Windows **Device Manager**](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Select **View > Devices by type** from the **Device Manager** menus.
1. Open the "**View**" menu.
1. If there is a **✓** to the left of the "**Show hidden devices**" menu item, click on "**Show hidden devices**" to disable it.
1. Open the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. You should see a port identified as "**USB-SERIAL CH340 (COM<i>n</i>)**" under the "**Ports (COM & LPT)**" section (where "COM<i>n</i>" is some serial port e.g., COM4). Double click on that item.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Driver**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
1. You will see a "**Roll Back Driver**" button in the dialog. If it is not clickable, perform the following instructions:
   1. Click the "**Update driver**" button.
      An "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog will open.
   1. Click on "**Search automatically for drivers**" in the "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog.
   1. You should now see the driver installation wizard update the driver. Wait for the update to finish, as indicated by the message "**Windows has successfully updated your drivers**" in the dialog.
      ❗ It is possible you will instead see the message "**The best drivers are already installed**". If so, please stop following the instructions here and comment on the issue to let me know. I'll provide alternative instructions you can follow.
   1. Click the "**Close**" button in the "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog.
      The dialog will close.
1. Click the "**Roll Back Driver**" button in the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
   The "**Driver Package rollback**" dialog will open.
1. Click the radio button next to "**My apps don't work with this driver**" in the "**Driver Package rollback**" dialog .
1. Click the "**Yes**" button.
   The "**Driver Package rollback**" dialog will close.
1. Click the "**Close**" button in the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.

Now start Arduino IDE again, upload the "hello" test sketch to your board. Hopefully it will no longer cause a BSoD.

When you perform the special "roll back" procedure per the instructions I provided above, Windows is smart enough to understand that it should not later update the driver to the problematic version `3.9.2024.9` again. So this fix should be persistent (until such time as WCH releases a new version of the driver, which historically they have only done every year or two, and hopefully might contain a fix anyway).

william-orange, 2025-01-13:
I have experienced the same problem as described here.  Going one version back on the serial driver did not help.  Slowing the baud rate to 9600 did not help.  I am presently using Putty for serial monitoring.

per1234, 2025-01-13:
Thanks for your report @william-orange. Which operating system are you using (e.g., "Windows 10")?

Which USB to serial chip does the board you are using have (e.g., "CH340")?

>I am presently using Putty for serial monitoring.

Does the BSoD also occur when using PuTTY, or does it only occur when using the Arduino IDE Serial Monitor?

william-orange, 2025-01-13:
I am using a variant of the Cheap Yellow Display.  I examined the serial chip with my jewler's lup.  Nothing is legible.  I believe it is a CH340 but I can not confirm it.
 
I am running windows 11 24h2 and the latest Arduini IDE release.  

The BSoD does not occure when using Putty.  I have let it run hours to be sure.

jtnorr, 2025-01-19:
Having a similar issue where either of the monitoring tools for data flow are causing BSOD. Otherwise the device can stay connected for as long as possible. I willl try updating the drivers but here are some details in the meantime:

USB to serial chipset: CH340C
Board: SparkFun RedBoard

nbourre, 2025-02-19:
I had the same in issue in my classroom which started in November. Here is [the link](https://forum.arduino.cc/t/sudden-issue-bsods-on-4-different-laptops-using-arduino-mega-ch341-in-classroom-today-first-time-in-3-months/1326498) to my post on the Arduino Forum.

Basically, I posted

> I teach basic robotics using Makeblock robots, which are essentially Arduino Mega boards.
> 
> Today, something unusual happened in my classroom. Four different laptops experienced BSODs (Blue Screens of Death) while working with the CH341 USB-to-serial chip. This is the first time we've encountered such an issue in three months of class.
> 
> Here are the details:
> 
>     The laptops are owned by students, and each is a different brand and model.
>     All the laptops are running Windows 11.
>     This issue occurred suddenly today, without any prior incidents in the past three months.
> 
> Has anyone else experienced similar issues with the CH341 chip, especially on Windows 11?

Wicz3k, 2025-02-20:
I also have this issue

alexew0923, 2025-03-06:
>        ❗ It is possible you will instead see the message "**The best drivers are already installed**". If so, please stop following the instructions here and comment on the issue to let me know. I'll provide alternative instructions you can follow.
My laptop's device manager says this, what should I do?

per1234, 2025-03-06:
Hi @alexew0923. I have found that when the driver installation gets into this state, it can be solved by uninstalling and then reinstalling the drivers. After that, the "roll back" option should be enabled.

I'll provide instructions you can follow to do that:

1. Close Arduino IDE if it is running.
1. Connect the Arduino board to your computer with a USB cable.
1. Open the [Windows **Device Manager**](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Select **View > Devices by type** from the **Device Manager** menus.
1. Open the "**View**" menu.
1. If there is a **✓** to the left of the "**Show hidden devices**" menu item, click on "**Show hidden devices**" to disable it.
1. Open the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. You should see a port identified as "**USB-SERIAL CH340 (COM<i>n</i>)**" under the "**Ports (COM & LPT)**" section (where "COM<i>n</i>" is some serial port e.g., COM4). Double click on that item.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Driver**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
1. Click the "**Uninstall Device**" button.
   The "**Uninstall Device**" dialog will open.
1. Check the box next to "**☐ Attempt to remove the driver for this device.**".
1. Click the "**Uninstall**" button.
   The "**Uninstall Device**" and "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialogs will close.
1. Disconnect the USB cable of the Arduino board from your computer.
1. Reconnect the USB cable of the Arduino board to your computer.
1. Wait for the "**USB-SERIAL CH340 (COM<i>n</i>)**" device to appear under the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
   ❗ This will take a couple of minutes as Windows must first automatically reinstall the driver.
   ❗ It might appear first as a "**USB2.0-Serial**" device under the "**Other devices** section of the tree. Just wait for a little longer and it will appear in its final place under the "**Ports (COM & LPT)**" section.
1. Click the following link to download the previous version of the driver from the chip manufacturer's website:
   <https://www.wch-ic.com/downloads/file/65.html?time=2023-03-16%2022:57:59>
1. Wait for the download to finish.
1. Run the downloaded `CH341SER.EXE` file.
1. A "**User Account Control**" dialog may now appear asking "**Do you want to allow this app to make changes to your device?**". Click the "**Yes**" button.
1. A "**DriverSetup(X64)**" dialog will open. Click the "**INSTALL**" button in the dialog.
1. Wait for the driver installation to finish, as indicated by the appearance of a "**Driver install success!**" dialog.
1. Click the "**OK**" button in the "**Driver install success!**" dialog.
   The dialog will close.
1. Click the **X** icon at the top right of the "**DriverSetup(X64)**" dialog.
   The dialog will close.
1. Switch back to the Windows "**Device Manager**".
1. Double click on the "**USB-SERIAL CH340 (COM<i>n</i>)**" device under the "**Ports (COM & LPT)**" section.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Driver**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
1. Click the "**Update driver**" button.
  An "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog will open.
1. Click on "**Search automatically for drivers**" in the "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog.
1. You should now see the driver installation wizard update the driver. Wait for the update to finish, as indicated by the message "**Windows has successfully updated your drivers**" in the dialog.
1. Click the "**Close**" button in the "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog.
  The dialog will close.
1. Click the "**Roll Back Driver**" button in the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
   The "**Driver Package rollback**" dialog will open.
1. Click the radio button next to "**My apps don't work with this driver**" in the "**Driver Package rollback**" dialog .
1. Click the "**Yes**" button.
   The "**Driver Package rollback**" dialog will close.
1. Click the "**Close**" button in the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.

Now start Arduino IDE again and upload the "hello" test sketch to your board. Hopefully it will no longer cause a BSoD.

alexew0923, 2025-03-07:
@per1234 I could fix the problem by following your steps, thank you so much!

per1234, 2025-04-04:
Hello again @Joko-1, @AkhmadKhoirudin, @bankdaysun, @william-orange, @jtnorr, @nbourre, @Wicz3k, and @alexew0923. A potential fix has been made in the [**serial-monitor**](https://github.com/arduino/serial-monitor) helper tool that provides communication between the Arduino IDE Serial Monitor tool and the serial port of the Arduino board:

https://github.com/bugst/go-serial/pull/200 / https://github.com/arduino/serial-monitor/pull/114

Arduino IDE automatically updates the **serial-monitor** when a new version is available, so the version with the fix (0.15.0) should already be installed on your computer and in use by Arduino IDE.

I already verified that the change another problem that affected boards with the CH340 chip (https://github.com/arduino/arduino-create-agent/issues/1000). However, I was never able to reproduce the problem reported here, so I am not able to check whether it is fixed or not.

Please check to see if the problem occurs when using the latest driver version and then add a comment here on the issue thread to let me know what you found.

I'll provide instructions you can follow to do that:

1. Connect the Arduino board to your computer with a USB cable.
1. Open the [Windows **Device Manager**](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Select **View > Devices by type** from the **Device Manager** menus.
1. Open the "**View**" menu.
1. If there is a **✓** to the left of the "**Show hidden devices**" menu item, click on "**Show hidden devices**" to disable it.
1. Open the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. You should see a port identified as "**USB-SERIAL CH340 (COM<i>n</i>)**" under the "**Ports (COM & LPT)**" section (where "COM<i>n</i>" is some serial port e.g., COM4). Double click on that item.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Driver**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
1. Click the "**Update driver**" button.
   An "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog will open.
1. Click on "**Search automatically for drivers**" in the "**Update Drivers Device - USB-SERIAL CH340 (COM<i>n</i>)**" dialog.
1. You should now see the driver installation wizard update the driver. Wait for the update to finish, as indicated by the message "**Windows has successfully updated your drivers**" in the dialog.
1. Click the "**Close**" button in the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.
1. Close Windows Device Manager.
1. Start Arduino IDE.
1. Upload the "hello" test sketch to the board:
   ```cpp
   void setup() {
     Serial.begin(9600);
   }
   void loop() {
     Serial.println("hello");
     delay(5);
   }
   ```
1. Open Serial Monitor.

Now start Arduino IDE again, upload the "hello" test sketch that I provided  to your board. Hopefully it will no longer cause a BSoD.

alexew0923, 2025-04-05:
> * You should now see the driver installation wizard update the driver. Wait for the update to finish, as indicated by the message "**Windows has successfully updated your drivers**" in the dialog.

When I do this, I get a message that says 'The best drivers for your device are already installed," instead of the one you told.

per1234, 2025-04-07:
Hi @alexew0923. I'm going to ask you to provide some additional information that might help us to understand how to proceed. Please do this:

1. Connect the Arduino board to your computer with a USB cable.
1. Open the [Windows **Device Manager**](https://support.microsoft.com/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Select **View > Devices by type** from the **Device Manager** menus.
1. Open the "**View**" menu.
1. If there is a **✓** to the left of the "**Show hidden devices**" menu item, click on "**Show hidden devices**" to disable it.
1. Open the "**Ports (COM & LPT)**" section of the **Device Manager** tree.
1. You should see a port identified as "**USB-SERIAL CH340 (COM<i>n</i>)**" under the "**Ports (COM & LPT)**" section (where "COM<i>n</i>" is some serial port e.g., COM4). Double click on that item.
   The "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog will open.
1. Select the "**Drivers**" tab of the "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.

Now please add a comment here on the issue thread to tell us the version number you saw after "**Driver Version**" field in that "**USB-SERIAL CH340 (COM<i>n</i>) Properties**" dialog.

alexew0923, 2025-04-08:
The driver version is 3.9.2024.9

alexew0923, 2025-04-09:
It looks like there is no more BSoD. Thanks a lot for helping us!

per1234, 2025-04-09:
Thanks for the update @alexew0923! Since the feedback I received from affected users here and on Arduino Forum indicates that the bug has indeed been fixed, I'm going to close this as resolved by https://github.com/bugst/go-serial/pull/200 and https://github.com/arduino/serial-monitor/pull/114.

If anyone still encounters a BSoD, please comment and I'll investigate further.

closed 2025-04-09 by per1234