arduino/arduino-ide #2593: Problem with chrome-sandbox in version 2.3.4.

gatherinegod, 2024-12-05:
Sorry but my english is a little bit poor.
I founded an error in chrome-sandbox. I'm in Lubuntu and I'm using the .zip IDE. I don't know what's the real problem but IDE executes but don't charge, only appears the load window but I can't use it. I replaced the chrome-sandbox file for the 2.3.3 version and it works now. 

This message is to warn about that error, I solved it.

per1234, 2024-12-05:
Thanks for your report @gatherinegod. I'm going to ask you to provide some additional information that might help us to identify the problem.

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Replace the `chrome-sandbox` file with the original file (or make a fresh installation of Arduino IDE 2.3.4).
1. Open [a command line terminal](https://ubuntu.com/tutorials/command-line-for-beginners) in the folder that contains the Arduino IDE installation.
1. Type the following command:
   ```text
   ./arduino-ide
   ```
1. Press the <kbd>**Enter**</kbd> key.
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Use the mouse to select all the text in the terminal window. Make sure to scroll all the way up to the top.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**C**</kbd> keyboard shortcut to copy the selected text to the clipboard.
1. Paste (<kbd>**Ctrl**</kbd>+<kbd>**V**</kbd>) the copied output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

gatherinegod, 2024-12-05:
Ok, first of all I have this problem, I need to configurate chrome-sandbox:

[23396:1205/191936.218932:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /home/drakter/Descargas/arduino-ide_2.3.4_Linux_64bit/chrome-sandbox is owned by root and has mode 4755.

That I solve doing this:
sudo chown root:root /path/to/chrome-sandbox
sudo chmod 4755 /path/to/chrome-sandbox

Now I don't know what's happening but works, maybe is because I used it before in the new version and all packages are updated?

closed 2024-12-05 by per1234