arduino/arduino-ide #2594: std::array seems broken

alanv72, 2024-12-06:
### Describe the problem

#include <vector>
#include <array>

struct ModbusData {
    uint16_t address;
    const char* name;
    float value;
    const char* unit;
    char* description;
    bool rw; // Read/Write capability
    bool rt; // Real-time update
};

#define MODBUS_DATA(addr, n, u, d, r, t) {addr, n, NAN, u, (char*)d, r, t}

static std::array<ModbusData, 29> modbusData = {
    MODBUS_DATA(0x100, "Battery SOC", "%", "Battery State of Charge", false, true),
    MODBUS_DATA(0x101, "Battery Voltage", "V", "Battery Voltage", false, true),
    MODBUS_DATA(0x102, "Battery Current", "A", "Battery Current", false, true),
    MODBUS_DATA(0x107, "PV1 Voltage", "V", "PV panel 1 voltage", false, true),
    MODBUS_DATA(0x108, "PV1 Current", "A", "PV panel 1 current", false, true),
    MODBUS_DATA(0x109, "PV1 Power", "W", "PV panel 1 power", false, true),
    MODBUS_DATA(0x10B, "Charge State", "-", "Charge state", false, false),
    MODBUS_DATA(0x10E, "Max Charge Power", "W", "Total charge power, include charge power by mains and PV", false, false),
    MODBUS_DATA(0x210, "Machine State", "-", "Machine state", false, false),
    MODBUS_DATA(0x213, "Grid Voltage A", "V", "Mains voltage phase A", false, true),
    MODBUS_DATA(0x214, "Grid Current A", "A", "Mains side input current phase A", false, true),
    MODBUS_DATA(0x215, "Grid Frequency", "Hz", "Mains frequency", false, true),
    MODBUS_DATA(0x216, "Inverter Voltage A", "V", "Inverter output voltage phase A", false, true),
    MODBUS_DATA(0x217, "Inverter Current A", "A", "Inverter inductive current phase A", false, true),
    MODBUS_DATA(0x218, "Inverter Frequency", "Hz", "Inverter frequency", false, true),
    MODBUS_DATA(0x219, "Load Current A", "A", "Load side current phase A", false, true),
    MODBUS_DATA(0x21B, "Load Active Power A", "W", "Load active power phase A", false, true),
    MODBUS_DATA(0x21E, "Mains Charge Current", "A", "Battery side current when charging on mains", false, true),
    MODBUS_DATA(0x21F, "Load Ratio A", "%", "Load percentage phase A", false, true),
    MODBUS_DATA(0x220, "Heat Sink A Temp", "℃", "DC-DC heat sink temperature", false, true),
    MODBUS_DATA(0x221, "Heat Sink B Temp", "℃", "DC-AC heat sink temperature", false, true),
    MODBUS_DATA(0x222, "Transformer temperature", "℃", "Transformer temperature", false, true),
    MODBUS_DATA(0x223, "Environment temperature", "℃", "Environment temperature", false, true),
    MODBUS_DATA(0x224, "PV Charge Current", "A", "Battery side current by PV charging", false, true),
    MODBUS_DATA(0xE001, "PV Charge Current Set", "A", "Set Max PV Charger Current in Amps", true, false),
    MODBUS_DATA(0xE205, "Grid Charge Current Limit", "A", "Maximum Grid charge current (0-40A)", true, false),
    MODBUS_DATA(0xE20A, "Total Max Charge Limit", "A", "Total Maximum Charge Current Limit", true, false),
    MODBUS_DATA(0x021B, "Inverter Load A", "W", "Inverter Load A", false, true),
    MODBUS_DATA(0xE204, "Output Priority", "-", "Output Priority: solar, line, sbu", true, false)
};

Compilation error: too many initializers for 'std::array<ModbusData, 29>'
---------------------------------------------------------------------------------------
struct ModbusData {
    uint16_t address;
//    const char* name;
//   float value;
//    const char* unit;
//    char* description;
//    bool rw; // Read/Write capability
//    bool rt; // Real-time update
};

//#define MODBUS_DATA(addr, n, u, d, w, t) {addr, n, NAN, u, (char*)d, w, t}

static std::array<ModbusData, 2> modbusData = {
    {0x100},
    {0x101}
};

Compilation error: too many initializers for 'std::array<ModbusData, 2>'
------------------------------------------------------------------------------------
struct ModbusData {
    uint16_t address;
};

ModbusData modbusData[2] = {
    {0x100},
    {0x101}
};

Compiles

Same data in a vector works.

### To reproduce

Arduino IDE 2.3.4

ex:

struct ModbusData {
    uint16_t address;
};

static std::array<ModbusData, 2> modbusData = {
    {0x100},
    {0x101}
};

### Expected behavior

Array gets initialized correctly and interacts correctly with macro expansion.

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

11 Pro 24H2

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2024-12-06:
Hi @alanv72. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the Arduino IDE application codebase hosted in this repository. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/programming-questions/20

closed 2024-12-06 by per1234

alanv72, 2024-12-06:
10-4. I'll find the appropriate place to open this issue.