arduino/arduino-ide #2602: ash fails when Windows username contains spaces

Rybec, 2024-12-20:
### Describe the problem

ash fails with the following error.  My Windows username has two spaces in it (thank Microsoft's garbage Windows installer for that).  I get the following error when attempting to build.  ("Thalin" is the middle portion of the username, and it has a space before and after it).

```
ash: Thalin: unknown operand
ash: Thalin: unknown operand
sdcpp.exe: fatal error: too many input files
compilation terminated.
subprocess error 1
ash: Thalin: unknown operand
ash: Thalin: unknown operand

exit status 1

Compilation error: exit status 1
```

### To reproduce

Install Windows 10, and give it a full name including middle, and it defaults to trying to just use the whole thing, spaces and all, as your username.
(TL;DR: Have two spaces in your Windows username.)
Install Arduino IDE (version 2.3.4 is what I'm using, but I did also test the nightly build noted in the version field).

I don't know how much difference this part makes, but I'm using ch55xduino with a QT Py CH552.  So I installed ch55xduino, setup the board, and so on.  (Additional notes on this later.)

Attempt to compile a simple sketch.  (Copy/pasted from Adafruit's QT Py CH552 tutorial.)

### Expected behavior

When I click "Verify" the sketch compiles successfully.

### Arduino IDE version

2.3.5-nightly-20241212 (please make About dialog copyable!)

### Operating system

Windows

### Operating system version

10

### Additional context

Note that the current official version of SDCC also has it's own spaces-in-path bug, and the SDCC that comes with c55xduino is old enough to still have this bug.  I have replaced that broken version of SDCC in my local copy of c55xduino with the most recent snapshot, which no longer has that bug (I've tested it at the command line).  The reason sdcpp.exe is failing is that it is getting an unquoted path with spaces as an argument, which ash interpreted as two separate arguments, because whatever is calling _it_ has failed to correctly quote the path.

For completeness, I've enabled verbose compilation.  Here is the output from that.  (Given that the initial error is thrown by ash, I doubt the wrapper shell script has anything to do with this, though it may create additional errors once the ash error is resolved.  I'll take a look at that tomorrow, and if it strips quotes without replacing them, I'll file a bug report with ch55xduino.  I already need to file a bug report for that project encouraging updating to the latest SDCC snapshot.)

Hmm, it looks like the problem may be with busybox, given that the compile command does seem have everything quoted...  ([According to MS](https://learn.microsoft.com/it-it/archive/blogs/twistylittlepassagesallalike/everyone-quotes-command-line-arguments-the-wrong-way), some terminal programs don't handle argument quoting correctly.  Maybe busybox is an example of this...)

```
FQBN: CH55xDuino:mcs51:ch552:clock=16internal
Using board 'ch552' from platform in folder: C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25
Using core 'ch55xduino' from platform in folder: C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25

Detecting libraries used...
C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.10/win/busybox ash C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.10/wrapper/sdcc.sh C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.13407_4/bin/sdcc C:\Users\Rybec Thalin Arethda\AppData\Local\arduino\sketches\F28B92F05926EE5B7BA4053F3A9A7367\sketch\sketch_dec20a.ino.cpp nul re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --model-large --int-long-reent -E -MC -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10607 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\cores\ch55xduino -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\variants\ch552 -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.13407_4/include
Generating function prototypes...
C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.10/win/busybox ash C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.10/wrapper/sdcc.sh C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.13407_4/bin/sdcc C:\Users\Rybec Thalin Arethda\AppData\Local\arduino\sketches\F28B92F05926EE5B7BA4053F3A9A7367\sketch\sketch_dec20a.ino.cpp C:\Users\Rybec Thalin Arethda\AppData\Local\Temp\1859949367\sketch_merged.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --model-large --int-long-reent -E -MC -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10607 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\cores\ch55xduino -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\variants\ch552 -IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.13407_4/include
ash: Thalin: unknown operand
ash: Thalin: unknown operand
C:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\Rybec Thalin Arethda\AppData\Local\Temp\1859949367\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\tools\\MCS51Tools\\2023.10.10/win/busybox" ash "C:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\tools\\MCS51Tools\\2023.10.10/wrapper/sdcc.sh" "C:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\tools\\sdcc\\build.13407_4/bin/sdcc" "C:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\arduino\\sketches\\F28B92F05926EE5B7BA4053F3A9A7367\\sketch\\sketch_dec20a.ino.cpp" "C:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\arduino\\sketches\\F28B92F05926EE5B7BA4053F3A9A7367\\sketch\\sketch_dec20a.ino.cpp.o" re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --model-large --int-long-reent --less-pedantic -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10607 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 "-IC:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\hardware\\mcs51\\0.0.25\\cores\\ch55xduino" "-IC:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\hardware\\mcs51\\0.0.25\\variants\\ch552" "-IC:\\Users\\Rybec Thalin Arethda\\AppData\\Local\\Arduino15\\packages\\CH55xDuino\\tools\\sdcc\\build.13407_4/include"
sdcpp.exe: fatal error: too many input files
compilation terminated.
subprocess error 1
ash: Thalin: unknown operand
ash: Thalin: unknown operand

exit status 1

Compilation error: exit status 1
```

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

Rybec, 2024-12-20:
Looking over the verbose output again, I can see the problem very clearly, and it's not busybox.  The busybox commands have _no_ quoting for any of the arguments.  So busybox is just passing them to ash as-is, and the problem is happening at the point where those busybox command are being generated.  The following ctags command does not have any quoting either.  Not sure if it is misbehaving as a result of that, but if it is, it's doing it silently.

Rybec, 2024-12-20:
Something else worth noting: SDCC cannot take multiple C files.  It can take a C file and multiple rel files (SDCC object files).  I'm not sure what the "re2" in the SDCC command is, but maybe it is being interpreted as an another file?

The two ash errors are almost certainly due to lack of any quoting at all in the first two busybox commands that invoke ash.  I wonder if the "re2" is an artifact of those failures?

I'll take a look at the wrapper script tomorrow and see exactly what it is doing with the arguments.  This might be related to [https://github.com/arduino/arduino-cli/issues/958](https://github.com/arduino/arduino-cli/issues/958), though that is a batch file and not a shell script.

per1234, 2024-12-20:
Hi @Rybec. The problem is a lack of quoting in the compilation command templates defined in the ch55xduino boards platform:

https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/platform.txt

Please report the bug to that project's issue tracker:

https://github.com/DeqingSun/ch55xduino/issues

> This might be related to https://github.com/arduino/arduino-cli/issues/958, though that is a batch file and not a shell script.

Nice work on finding that issue! However, as you pointed out, that bug is specific to batch files so it is not relevant. This is actually good news because working around that bug is difficult, while in this case the fix is simply a matter of adding some quotes in `platform.txt`.

closed 2024-12-20 by per1234

Rybec, 2025-01-03:
Related to: https://github.com/DeqingSun/ch55xduino/issues/175

Ok, this might indeed be an Arduino issue.  I've looked at the various files, I've worked out what commands are actually being executed, I've messed with the shell script file, and I've experimented at the command line.  Here's what I've worked out:

The ash errors seem to be separate from the sdcpp error.  Those are part of the  "Detecting libraries used..." and "Generating function prototypes..." steps, and they aren't using the quoting that appears to be specified in platform.txt.  (Can someone with more experience look at platform.txt in ch55xduino and let me know if the quoting looks right?  While things appear to be correctly quoted, I don't actually know exactly what it should look like...  If there are things in there that are not quoted correctly, please let me know what correct quoting looks like, and I'll work with DeqingSun to fix it.)

As far as I can tell, the wrapper shell script is quoting arguments correctly, _except_ that certain arguments are being passed to the script by Arduino without proper quoting.  The paths in question are the include paths:

`-IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\cores\ch55xduino`
`-IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.25\variants\ch552`
`-IC:\Users\Rybec Thalin Arethda\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.13407_4/include`

I'm attempting to modify the wrapper script to quote these, but it' a bit of a pain, because quoting 100% of the arguments fails (because some arguments don't like to be quoted), so I'm going to have to look for `-I` arguments specifically and only quote those.  Is there a better way of fixing this?  I know next to nothing about how Arduino constructs these commands, so I'm kind of flying by the seat of my pants here.

Solutions I've found that allow compilation when working from the command line include quoting the full argument, eg. `"-Ifile path with spaces"`, quoting just the file path, eg. `-I"file path with spaces"`, and separating the `-I` from the file path and quoting the file path, eg. `-I "file path with spaces"`.  I assume that any of these would be acceptable within the script, however the argument is passed to the script as a single argument with no quoting, eg. `-Ifile path with spaces`.  I'm sure it wouldn't be too hard to split them up in the shell script, but my shell scripting skills are pretty basic, so it would take me a significant amount of time to learn everything necessary and then apply the changes myself.  If there's a better option I would very much prefer to do that.

Rybec, 2025-01-03:
Ok, they are properly quoted coming into the script, as they enter as complete arguments rather than individual components.  Is there a way in platform.txt to add a second layer of quoting to just the `-I` arguments, so that there is still one layer of quotes remaining after entering the shell script?

If not, then it looks like the only solution is for me to learn shell scripting better and add quotes back on in the script.

Rybec, 2025-01-03:
Line 76 in platform.txt might be a place where an additional layer of quotes (probably escaped) would help, however that will only solve one of the three include lines (the one that ends with `/include`).  I'm not seeing a place where formatting for the others can be changed.  (They seem to be injected by Arduino, under the name `includes` in platform.txt.)

Rybec, 2025-01-03:
https://github.com/DeqingSun/ch55xduino/issues/175#issuecomment-2569927692

I tried this.  It appears to work for the third `-I` argument, but it doesn't affect the ones in `include` passed in by Arduino.