arduino/arduino-ide #2604: Offline Network Warnings

Timfedora, 2024-12-22:
### Describe the request

Hi, everyone. I get a warning popup every time I open the IDE, when it's not connected to the internet. I was wondering if there is a setting I could change to prevent this.

### Describe the current behavior

I have installed the IDE using flatpak.
I have tried disabling the automatic updates as posted in #2401
I have not verified the feature was still missing when using the latest [nightly build]

### Arduino IDE version

2.3.4

### Operating system

Linux

### Operating system version

Fedora 41

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

Timfedora, 2024-12-22:
[image]

pogo555, 2025-04-08:
Similar, but also at odd times while using the IDE. If I'm hundreds of kilometres from the internet (as happens often enough), I don't need to be reminded. In fact I'd rather not know. Can this warning have a checkbox to turn it off?