arduino/arduino-ide #2607: IDE should automatically pick up changes made to files in external editors

JetForMe, 2025-01-04:
### Describe the request

The IDE should automatically pick up changes made to files from outside the IDE.

In order to more rapidly prototype some code, I develop it in Xcode, and then periodically build in Arduino to see how it works on the device. Every time I make changes externally, the Arduino editor fails to notice, and when I save changes in the IDE, it says something like "The file <> has been changed on the file system, do you want to overwrite it." I almost never want to overwrite it, but I have to answer this for every file that might've changed.

Furthermore, the IDE doesn't see the changes on disk, and continues to show stale contents in the editor tabs. It should refresh these contents so I can seamlessly edit in both environments.

Many editors have a preference to do this, but I couldn't find one in Arduino IDE.

There have been long-standing similar requests: https://github.com/arduino/Arduino/issues/4551

### Describe the current behavior

In order to more rapidly prototype some code, I develop it in Xcode, and then periodically build in Arduino to see how it works on the device. Every time I make changes externally, the Arduino editor fails to notice, and when I save changes in the IDE, it says something like "The file <> has been changed on the file system, do you want to overwrite it." I almost never want to overwrite it, but I have to answer this for every file that might've changed.

Furthermore, the IDE doesn't see the changes on disk, and continues to show stale contents in the editor tabs. It should refresh these contents so I can seamlessly edit in both environments.

### Arduino IDE version

In order to more rapidly prototype some code, I develop it in Xcode, and then periodically build in Arduino to see how it works on the device. Every time I make changes externally, the Arduino editor fails to notice, and when I save changes in the IDE, it says something like "The file <> has been changed on the file system, do you want to overwrite it." I almost never want to overwrite it, but I have to answer this for every file that might've changed.  Furthermore, the IDE doesn't see the changes on disk, and continues to show stale contents in the editor tabs. It should refresh these contents so I can seamlessly edit in both environments.

### Operating system

macOS

### Operating system version

15.2

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

ftlpilot, 2025-01-04:
IMHO, disabling the internal Editor to use an EXTERNAL EDITOR should BE AN OPTION, as was the case in earlier versions of the IDE.

Currently, accidentally adding text to the internal editor STOPS SYNCING WITH THE (DESIRED) EXTERNAL EDITOR.

Until the IDE allows MULTIPLE WINDOWS, it just a toy editor

per1234, 2025-01-07:
Hi @JetForMe. Arduino IDE already does this. It will only pick up the changes to the file on disk if the sketch is in a saved state. This is an intentional design in order to avoid the loss of your unsaved work that is staged in the Arduino IDE editor.

You can ensure the sketch is always in a saved state by enabling the "Auto save" preference. The IDE will then save the sketch automatically whenever you made any edits via the Arduino IDE editor. Of course this setting significantly changes the behavior of Arduino IDE, so you should make sure to consider whether this is something you want, and to adjust your workflow as needed.

closed 2025-01-07 by per1234

ftlpilot, 2025-01-07:
> Hi [@JetForMe](https://github.com/JetForMe). Arduino IDE already does this. It will only pick up the changes to the file on disk if the sketch is in a saved state. This is an intentional design in [...]

As seems usual - YOU COMPLETELY MISSED THE POINT I WAS MAKING.
FIRSTLY - I NEVER, EVER want to use the Arduino IDE INTERNAL EDITOR and ALWAYS USE AN EXTERNAL (Multi-Window) Editor.
Therefore, the IDE refusing to Sync - WHILE HIDING THE FACT IS ISN'T SYNCING has WASTED A LOT OF MY TIME.

The solution is easy.
-- just --
ADD OPTION TO DISABLE INTERNAL EDITOR

ftlpilot, 2025-01-07:
> Closed #2607 <https://github.com/arduino/arduino-ide/issues/2607>  as not planned.
>
>
> This email has been checked for viruses by Avast antivirus software.
> www.avast.com

As seems usual - YOU COMPLETELY MISSED THE POINT I WAS MAKING.

FIRSTLY - I NEVER, EVER want to use the Arduino IDE INTERNAL EDITOR and ALWAYS USE AN EXTERNAL (Multi-Window) Editor.

Therefore, the IDE refusing to Sync - WHILE HIDING THE FACT IS ISN'T SYNCING has WASTED A LOT OF MY TIME.

 

The solution is easy.

-- just --

ADD OPTION TO DISABLE INTERNAL EDITOR