arduino/arduino-ide #2608: UNPLUGGING ARDUINO BOARD loses Board Type - EVEN THOUGH IDENTICAL BOARD IS plugged back in

ftlpilot, 2025-01-04:
### Describe the problem

A very annoying BUG is the way the IDE loses the board type when its unplugged - even if it (or an identical one) is plugged back in immediately. This should be a selectable option

### To reproduce

just read above

### Expected behavior

.

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

10

### Additional context

.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-01-07:
Thanks for your report @ftlpilot. I am not able to reproduce the issue, and have not seen reports from other users of what sounds like a very impactful bug.

I'm not sure I understood the problem due to the lack of details in your report. Please add a comment here to provide more details, including:

- Which specific board or boards are you using when this occurs? If you are using a 3rd party derivative or clone board, and the board has a dedicated USB to serial bridge chip, tell us which specific chip it is (e.g., Microchip ATmega16U2, WCH CH340, Silicon Labs CP2102).
- What exactly do you mean by "loses board type".
  - Are you referring to the identification the board selector menu on the IDE toolbar?:
    [image]
  - Or are you referring to the selection in the **Tools > Board** menu?:
    [image]
  - Or are you referring to something else?

per1234, 2025-01-15:
Closing due to lack of response.

@ftlpilot if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2025-01-15 by per1234