arduino/arduino-ide #2609: REQUEST: automatically skip "verify" if already verified and no additional changes

jmmon, 2025-01-05:
### Describe the request

Hi, I see a recent update allows skipping verification during upload. I don't want to use this because it's nice to verify. But it led me to an idea: if it's already been verified but it failed to upload, then when I try again to upload there is no need to verify a second time (unless changes were made in any of the files).

Two possible implementations: 
general:
- e.g. WHENEVER verifying a sketch, check if it was already declared as "valid/verified" and also check that there were no changes to any files. 
- In this case, skip verification and show a notification "This sketch is (or "has already been") verified successfully." or more verbose: "No changes were made and sketch is already successfully been verified. Skipping verification step."

upload-only:
- Same as above, but only applies to the "upload" step e.g. "verify then upload" case.
- e.g. when uploading, sometimes it fails to upload due to comms issue, but it did successfully go through the verification step before attempting upload.
- In this case, can display the same notification as above under "general" implementation, and skip the verification check and instead jump directly to attempting to upload the sketch.

### Describe the current behavior

Currently, whenever there is an upload failure and I attempt to re-upload a sketch, the IDE does both a "verification" and then attempts an "upload." (Reading patch notes,) There was a recent change to allow skipping of verification COMPLETELY during the upload phase. 

While nice, I won't use that feature because when I upload I typically do want it to verify the code in the case that it has not already been verified.

This change request would allow automatic optimization for uploads. Basically, only verify when needed (changes made to files && not already verified during this session). Else, skip verification step and go straight to uploading.

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

10

### Additional context

This could be implemented as an opt-in feature initially, but perhaps in the future (after battle-testing) it could be implemented as a default. I would think most people would prefer to save time by skipping unnecessary repeated-verification steps, so long as there are no file changes and the IDE knows the sketch is already valid from a previous verification (during the same session.) 

I think it's smart to make it verify at least once per session at least, just to be certain. I wouldn't a case where someone updates the IDE (or some other change,) and then it skips the verification and fails. That's why I suggest using the session as the limit for what counts as "previously verified." When opening a new window, or when loading a sketch, it would be considered a new session so it will have to run the verification step at least once. After that initial verification (and if no file changes) then it can skip additional verifications during upload. 

And one could still manually click the Verify button to run the verification, even if it's already considered "verified" by the IDE. So that functionality would remain the same.

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

per1234, 2025-01-07:
Thanks for taking the time to make this suggestion @jmmon. Significant effort has already been put into making the Arduino sketch build system avoid unnecessary compilations. That work will continue, but it is not so simple as you might think because we must consider the possibility of changes to any of the dependencies (libraries, platforms, or toolchain); not only to the sketch code. The code for determining such things is not in the Arduino IDE codebase, but instead in its Arduino CLI helper tool, so that work will not occur in this repository.

closed 2025-01-07 by per1234

jmmon, 2025-01-07:
Thank you for the info! I understand the complications. I appreciate the v2 IDE and all the work that's gone into it! Keep up the good work!