arduino/arduino-ide #2610: Arduino IDE v2.3.4 Linux Mint 22

lexmark3200, 2025-01-06:
### Describe the problem

After upgrade from v2.3.2 on Linux Mint it doesn't start anymore. After starting shows the window with aplication and the last file, but the window of aplication is white then a few seconds later is blinking.
I removed instalation, then install from APP Store on Linux Mint 22, the problem is still present. So I installed it from Warehouse, the same issue.
Aplication is starting properly only if it starting _.AppImage_, or if I remove "User data files" from ~/.var/app/cc.arduino.IDE2

[image]

### To reproduce

You can try to install older version, then make some settings, close and just upgrade to latest and try to run aplication. 

### Expected behavior

Starting properly without deleting config files.

### Arduino IDE version

2.3.4

### Operating system

Linux

### Operating system version

Mint 22

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-01-06:
Hi @lexmark3200. I believe this bug is specific to the 3rd party [Flatpak](https://flatpak.org/) package of Arduino IDE: https://github.com/flathub/cc.arduino.IDE2/issues/100. This is further evidenced by your statement:

> Aplication is starting properly only if it starting .AppImage

I assume the AppImage you are referring to is the official build distributed by Arduino.

I see you already found the fix of clearing the cache folder:

> remove "User data files" from ~/.var/app/cc.arduino.IDE2

So I will close this as off topic.

closed 2025-01-06 by per1234

lexmark3200, 2025-01-06:
Ok thx for your reply.
I doscovered the problem is with this Direcotry: _~.var/app/cc.arduino.IDE2/config/arduino-ide/GPUCache_
Just remove this directory nad aplication is starting properly.