arduino/arduino-ide #2613: Save on desktop when redirection folder is actived open new empty sketch

UNISTRAPHIOlivier, 2025-01-08:
### Describe the problem

Hello Dear all.

We are encountering a bug.
We are operating in a Windows 10 22H2 environment.
The computers are integrated into an Active Directory with a GPO for folder redirection on Documents and the Desktop.

When we save a sketch to C:\temp\sketch_test, for example, it works.
However, when we save it to the Desktop, a new window opens with an empty sketch.

When we save it to a network drive like U:\, it works, but when we use the UNC path, it does not work.

I assume that folder redirection and saving via UNC path have the same issue.

Here is the output from the Arduino window when saving to C:\temp (where it works):

 ```
Closing channel on service path '/services/electron-window'.
Closing channel on service path '/services/ide-updater'.
2025-01-08T17:10:07.366Z root ERROR Uncaught Exception:  Error: transport close
2025-01-08T17:10:07.366Z root ERROR Error: transport close
    at C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1253913
    at Map.forEach (<anonymous>)
    at C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1253892
    at C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1221089
    at i.invoke (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1221179)
    at s.fire (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1222982)
    at C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1248775
    at Map.forEach (<anonymous>)
    at Object.<anonymous> (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1248745)
    at e.dispose (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:1214148)
Opening channel for service path '/services/electron-window'.
Opening channel for service path '/services/ide-updater'.
2025-01-08T17:10:08.190Z root WARN A command editor.action.toggleStickyScroll is already registered.
2025-01-08T17:10:08.291Z root INFO Start frontend contributions: 140.1 ms [Finished 0.896 s after frontend start]
2025-01-08T17:10:08.291Z root INFO Changed application state from 'init' to 'started_contributions'.
2025-01-08T17:10:08.291Z root INFO Changed application state from 'started_contributions' to 'attached_shell'.
2025-01-08T17:10:08.291Z root INFO >>> Restoring the layout state...
2025-01-08T17:10:08.291Z root INFO ------- SERIALIZED LAYOUT DATA -------
2025-01-08T17:10:08.291Z root INFO {"version":5,"mainPanel":{"main":{"type":"tab-area","widgets":[{"constructionOptions":{"factoryId":"code-editor-opener","options":{"counter":0,"kind":"navigatable","uri":"file:///c%3A/temp/sketch_test/sketch_test.ino"}},"innerWidgetState":"{\"cursorState\":[{\"inSelectionMode\":false,\"selectionStart\":{\"lineNumber\":3,\"column\":5},\"position\":{\"lineNumber\":3,\"column\":5}}],\"viewState\":{\"scrollLeft\":0,\"firstPosition\":{\"lineNumber\":1,\"column\":1},\"firstPositionDeltaTop\":0},\"contributionsState\":{\"editor.contrib.folding\":{\"lineCount\":10,\"provider\":\"indent\",\"foldedImports\":false},\"editor.contrib.wordHighlighter\":false}}"}],"currentIndex":0}},"mainPanelPinned":[false],"bottomPanel":{"config":{"main":null},"pinned":[],"expanded":false},"leftPanel":{"type":"sidepanel","items":[{"widget":{"constructionOptions":{"factoryId":"arduino-sketchbook-widget"}},"rank":1,"expanded":false,"pinned":false},{"widget":{"constructionOptions":{"factoryId":"boards-list-widget"}},"rank":2,"expanded":false,"pinned":false},{"widget":{"constructionOptions":{"factoryId":"library-list-widget"}},"rank":3,"expanded":false,"pinned":false},{"widget":{"constructionOptions":{"factoryId":"debug"},"innerWidgetState":"{\"parts\":[{\"partId\":\"debug:threads:-1\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"debug:threads\"},\"innerWidgetState\":\"{}\"}},{\"partId\":\"debug:frames:-1\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"debug:frames\"},\"innerWidgetState\":\"{}\"}},{\"partId\":\"debug:variables:-1\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"debug:variables\"},\"innerWidgetState\":\"{}\"}},{\"partId\":\"debug:watch:-1\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"debug:watch\"},\"innerWidgetState\":\"{}\"}},{\"partId\":\"debug:breakpoints:-1\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"debug:breakpoints\"},\"innerWidgetState\":\"{}\"}},{\"partId\":\"plugin-view:cortex-debug.peripherals\",\"collapsed\":true,\"hidden\":true,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"plugin-view\",\"options\":{\"id\":\"plugin-view:cortex-debug.peripherals\",\"viewId\":\"cortex-debug.peripherals\"}},\"innerWidgetState\":\"{\\\"label\\\":\\\"Cortex Peripherals\\\",\\\"widgets\\\":[],\\\"suppressUpdateViewVisibility\\\":false,\\\"currentViewContainerId\\\":\\\"debug\\\"}\"}},{\"partId\":\"plugin-view:cortex-debug.registers\",\"collapsed\":true,\"hidden\":true,\"originalContainerId\":\"debug:view-container:-1\",\"originalContainerTitle\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"plugin-view\",\"options\":{\"id\":\"plugin-view:cortex-debug.registers\",\"viewId\":\"cortex-debug.registers\"}},\"innerWidgetState\":\"{\\\"label\\\":\\\"Cortex Registers\\\",\\\"widgets\\\":[],\\\"suppressUpdateViewVisibility\\\":false,\\\"currentViewContainerId\\\":\\\"debug\\\"}\"}}],\"title\":{\"label\":\"debug\",\"iconClass\":\"codicon codicon-debug-alt\",\"closeable\":true}}"},"rank":4,"expanded":false,"pinned":false},{"widget":{"constructionOptions":{"factoryId":"search-view-container"},"innerWidgetState":"{\"parts\":[{\"partId\":\"search-in-workspace\",\"collapsed\":false,\"hidden\":false,\"originalContainerId\":\"search-view-container\",\"originalContainerTitle\":{\"label\":\"Search\",\"iconClass\":\"codicon codicon-search\",\"closeable\":true},\"widget\":{\"constructionOptions\":{\"factoryId\":\"search-in-workspace\"},\"innerWidgetState\":\"{\\\"matchCaseState\\\":{\\\"className\\\":\\\"codicon codicon-case-sensitive\\\",\\\"enabled\\\":false,\\\"title\\\":\\\"Match Case\\\"},\\\"wholeWordState\\\":{\\\"className\\\":\\\"codicon codicon-whole-word\\\",\\\"enabled\\\":false,\\\"title\\\":\\\"Match Whole Word\\\"},\\\"regExpState\\\":{\\\"className\\\":\\\"codicon codicon-regex\\\",\\\"enabled\\\":false,\\\"title\\\":\\\"Use Regular Expression\\\"},\\\"includeIgnoredState\\\":{\\\"className\\\":\\\"codicon codicon-eye\\\",\\\"enabled\\\":false,\\\"title\\\":\\\"Include Ignored Files\\\"},\\\"showSearchDetails\\\":false,\\\"searchInWorkspaceOptions\\\":{\\\"matchCase\\\":false,\\\"matchWholeWord\\\":false,\\\"useRegExp\\\":false,\\\"includeIgnored\\\":false,\\\"include\\\":[],\\\"exclude\\\":[],\\\"maxResults\\\":2000},\\\"searchTerm\\\":\\\"\\\",\\\"replaceTerm\\\":\\\"\\\",\\\"showReplaceField\\\":false,\\\"searchHistoryState\\\":{\\\"history\\\":[],\\\"index\\\":0},\\\"replaceHistoryState\\\":{\\\"history\\\":[],\\\"index\\\":0},\\\"includeHistoryState\\\":{\\\"history\\\":[],\\\"index\\\":0},\\\"excludeHistoryState\\\":{\\\"history\\\":[],\\\"index\\\":0}}\"}}],\"title\":{\"label\":\"Search\",\"iconClass\":\"fa fa-arduino-search\",\"closeable\":true}}"},"rank":5,"expanded":false,"pinned":false}]},"rightPanel":{"type":"sidepanel","items":[]},"activeWidgetId":"code-editor-opener:file:///c%3A/temp/sketch_test/sketch_test.ino:0" [... cut ...]
  // put your setup code here, to run once:
test
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

When i tape ctrl+s instead of just saving, I get the save dialog box again

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

aliphys, 2026-10-21:
I think I am replicating this issue.

Examples open without problem. I cannot access Arduino sketchs from the quick access location which are stored on a mapped drive (context: testing course content at Lund University, need to check if students can easily access the Arduino IDE for their project).

Instead an empty sketch is opened up.

Workaround is to navigate to the shared drive directly, and opening the .ink from there.

aliphys, 2026-10-21:
Video demonstration of issue

https://github.com/user-attachments/assets/4666cb0f-5a2a-4b26-843d-8848fe894455