arduino/arduino-ide #2614: IDE Wants to install already installed dependencies

ropg, 2025-01-08:
### Describe the problem

I wrote a library called LVGL_CYD, it is in the index. It has in its library.properties a depends line that reads

```
depends=lvgl,TFT_eSPI,spi_lcd_read
```

The first two being what you think they are, the last one another small library I wrote, also in the index. When I have the dependencies already installed, the IDE nonetheless prompts:

[image]

When I hit "Install All", the output is:

[image]

### To reproduce

* Install LVGL_CYD and dependencies
* Uninstall LVGL_CYD
* Install LVGL_CYD

### Expected behavior

When dependencies already installed, just install the library and not prompt to install the dependencies.

### Arduino IDE version

2.3.4

### Operating system

macOS

### Operating system version

MacOS Sequoia 15.2 (but also occurs on Windows say library users)

### Additional context

I might well be doing something subtly wrong in that library, but I don't see it and cannot explain this behavior by the IDE.

Same problem on 2.3.5-nightly-20241212, which I found [here](https://forum.arduino.cc/t/nightly-builds-for-modern-macs/1328582). Is that the official download link for nightly builds for Apple silicon?

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-01-08:
Hi @ropg. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/721.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

ropg, 2025-01-08:
Understood. Hope it gets fixed at some point, given that it's been around for a while. For now I'll just tell my users to expect it.