arduino/arduino-ide #2615: python3 could not be found at compile time

LHX980818, 2025-01-14:
### Describe the problem

I use the esp32-cam,The following problem occurred at compile time:

[image]

My terminal can use python commands

[image]

### To reproduce

macos(15.1.1)、arduino2.1.1  \ arduino2.3.4、esp32-cam

#include <WiFi.h>
#include "esp_camera.h"

### Expected behavior

Compile and upload successfully

### Arduino IDE version

2.3.4

### Operating system

macOS

### Operating system version

15.1.1

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

LHX980818, 2025-01-14:
I configured the Arduino using the following command，Because Arduino can't find python

`sed -i -e 's/=python /=python3 /g' ~/Library/Arduino15/packages/esp32/hardware/esp32/*/platform.txt`

per1234, 2025-01-14:
Hi @LHX980818. The '`"python3": cannot run executable found relative to current directory`' error you encountered occurs under the following conditions:

- A command is invoked that does not specify the path to the executable
- The executable is present in the working directory of the command

This is done intentionally by the `exec` package of the Go programming language that Arduino IDE uses to invoke commands. It is a security measure:

https://pkg.go.dev/os/exec#hdr-Executables_in_the_current_directory

> will not resolve a program using an implicit or explicit path entry relative to the current directory. That is, if you run `exec.LookPath("go")`, it will not successfully return `./go` on Unix nor `.\go.exe` on Windows, no matter how the path is configured. Instead, if the usual path algorithms would result in that answer, these functions return an error err satisfying `errors.Is(err, ErrDot)`.

https://cs.opensource.google/go/go/+/master:src/os/exec/exec.go;l=1303

```go
var ErrDot = errors.New("cannot run executable found relative to current directory")
```

The ["**esp32**" boards platform](https://docs.espressif.com/projects/arduino-esp32/en/latest/getting_started.html) defines several `python3` commands without an explicit path to `python3`:

https://github.com/espressif/arduino-esp32/blob/3.1.1/platform.txt#L15-L24

This indicates that you are starting Arduino IDE from the folder that contains your `python3` executable. The simple solution is to start Arduino IDE from any other folder. If you put the `python3` executable in the Arduino IDE installation folder, you should move that executable to another location on your computer, and make sure that location is in your system PATH.

As I mentioned above, this general behavior of Arduino IDE is intentional. Note that Arduino IDE itself doesn't know anything about `python3`. It is only running the arbitrary command that was defined by the **esp32** boards platform.

So I will close this as off topic for this repository.

The command defined by the **esp32** platform is a bit unusual, since it depends on you having installed Python 3, and placed the location of the `python3` executable in your system path, whereas traditionally Arduino boards platforms have provided all their dependencies. But this is an intentional choice by the **esp32** platform developers, so I also don't think it would be appropriate to report it to them.

In the end, it is a matter of a misconfiguration of your system. It sounds like you have already found a solution of sorts, but if you would like any further assistance or discussion, you are welcome to post over on Arduino Forum:

https://forum.arduino.cc/c/development-tools/ide-2-x/93

closed 2025-01-14 by per1234