arduino/arduino-ide #2616: Incomplete Dutch translation in menu bar

sterretjeToo, 2025-01-15:
### Describe the problem

I just switched from using the IDE in English to using it in Dutch and noticed that `File` and 'Edit` are not translated.

`File` should be `Bestand`.
`Edit`should be `Bewerken`.

### To reproduce

Change the language to Dutch in file -> preferences.

### Expected behavior

Correct translations

### Arduino IDE version

2.3.4 and 2.3.5-nightly-20241212

### Operating system

Windows

### Operating system version

11 Home

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

sterretjeToo, 2025-01-16:
Found another issue; not sure if I have to create a new issue. In library manager, there is no space between the version number of the library and the "geïnstalleerd" (installed). In the English version there is this space.

[image]

per1234, 2025-04-11:
Thanks for your reports @sterretjeToo!

>`File` should be `Bestand`.

I verified that this bug still exists in 2.3.6.

>`Edit` should be `Bewerken`.

This has now been fixed. It is at least progress towards a resolution of this issue.

> In library manager, there is no space between the version number of the library and the "geïnstalleerd" (installed).

This was caused by a mistake made by the community translator:

https://github.com/arduino/arduino-ide/blob/0f9f0d07b7d5ff0cdefda3e77eb6b5ce2854c4a8/i18n/nl.json#L172

(note the lack of a space between the template placeholder `{0}` and the string `geïnstalleerd`. I have corrected the error in the Transifex project which is the source of this localization data. The fix will be pulled into this repository by the next sync of the localization data.

If you would like to contribute to the Dutch translations, it would be very welcome. This is done via the Transifex project here:

https://explore.transifex.com/arduino-1/ide2/

The localization infrastructure is very streamlined and mostly automated, so you can be confident that Arduino will be able to ship any translation contributions you make to the users by the next production release of Arduino IDE.

>mix of English and Dutch in the notification are when updating a library.

The reason for this is that the "Verwerken" string is defined in the Arduino IDE 2.x codebase, and so is localized via this project's infrastructure:

https://github.com/arduino/arduino-ide/blob/0f9f0d07b7d5ff0cdefda3e77eb6b5ce2854c4a8/arduino-ide-extension/src/browser/widgets/component-list/filterable-list-container.tsx#L148

https://github.com/arduino/arduino-ide/blob/0f9f0d07b7d5ff0cdefda3e77eb6b5ce2854c4a8/i18n/nl.json#L150

but the rest of the message is coming from Arduino CLI and Arduino IDE 2.x can only display that text as received. We do have significant localization of Arduino CLI as well, but surprisingly we don't have Dutch locale coverage:

https://explore.transifex.com/arduino-1/arduino-cli/

Arduino provides the internationalization capabilities, but localization is driven by the community. So if the community requests the addition of the Dutch locale to the Arduino CLI Transifex project, and then contributes translations for the strings of the Arduino CLI user interface, then the text that originates from Arduino CLI will automatically be localized to Dutch in the Arduino IDE UI, without any need for changes in the codebases or infrastructure of either project.