arduino/arduino-ide #2618: Serial output stops responding Ubuntu 20.04LTS

galinawind, 2025-01-21:
### Describe the problem

I recently upgraded to Ubuntu 20.04LTS. And immediately started having issues with the Arduino IDE. I was running 2.3.2. I upgraded to 2.3.4, thinking that this was the problem, but didn't help at all. The issue is as follows. The serial monitor output works for anything in the setup() subroutine. The serial monitor stops working for anything in loop(). I am doing a complicated operation set using a NeoKey 1x4 and I need to see that the buttons are registering callbacks correctly. Yesterday I thought that my hardware had failed. Took several hours I didn't have to figure out that it was the IDE. 

I would start the program. If I immediately hit one of the keys, it would output for some time. If I pause for even a tiny bit, the serial monitor freezes. If I navigate to a web browser or something, the serial monitor freezes. I can un-freeze it temporarily by randomly changing the baud rate from the drop-down menu. Any change at all, to anything at all, unfreezes the monitor, but again, only temporarily. Which is extremely annoying. The operation the button set has to perform is to start in an SD card browsing mode. Then when it is detected that the folder only contains MP3 files, the button set switches to audio playback control, with one button being an exit back to folder browsing. So you see why I'd want to have the serial output going at least until I am certain that everything is working the way it should be. At least for now, just turning off the serial monitor is not exactly an option. 

If there are any solutions to this, I haven't seen any. Please advise. 

### To reproduce

Connect Adafruit Feather RP2040 and NeoKey 1x4 with a StemmaQT cable. Plug the lot into the USB port. If you have a power switch wired to the assembly, even better as hot-plugging the thing is kind of eh. Compile and load the basic_callback sketch onto the RP2040. Push the buttons a couple of times so that you see the printouts. Navigate away from the IDE to a web browser, or any other window. Come back to the IDE and see that nothing is being printed anymore. Change the baud rate from the drop-down menu on the serial monitor. You will see that the printouts are appearing again. Stop interacting for like a count of 5. The serial monitor will freeze again. Change the baud rate to some random number from the menu. The interaction will pick up again for a short time. 

### Expected behavior

You will see that the serial monitor freezes and briefly resumes if you change the baud rate in the menu to some random value. 

### Arduino IDE version

2.3.2 and 2.3.4

### Operating system

Linux

### Operating system version

Ubuntu 20.04LTS

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-01-22:
Thanks for your report @galinawind. Please tell us whether the fault still occurs if you upload this simple sketch to the board:

```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("hello");
  delay(1000);
}
```

galinawind, 2025-01-22:
There is no fault if this simple sketch runs. To be exact, there is never any fault. There is never any error anywhere. The Serial monitor just stops working. When I was on Ubuntu 18.04, I was not having any issues whatsoever.

per1234, 2025-01-23:
>there is never any fault. There is never any error anywhere. The Serial monitor just stops working.

When I said "fault", I was referring to the problem you are reporting: 

> the serial monitor freezes

Now please provide the information I requested in my previous comment: After you upload the simplified sketch to your board, does the Serial Monitor freeze on your Linux computer like it does when the board is running the complex "basic_callback" sketch?

galinawind, 2025-01-23:
No, the serial monitor does not freeze when I run a simple sketch. 

I now have a work-around. Everything works perfectly if I use the Mac build of the 2.3.4 IDE. I don't like Macs and my Intel MacBook Pro is a piece of trash, but this one thing works on it. Hopefully you can fix the Linux build at some point.