arduino/arduino-ide #2620: 'Export Compiled Binary' Not working for ATMEGA32u4 boards

Carter3DP, 2025-01-27:
### Describe the problem

When using the 'Export compiled binary' option to get binaries for a selected sketch, the binary with both the program and the bootloader, labeled "sketch.ino.with_bootloader.hex", will only contain the bootloader.

Verbose Console Output:
`FQBN: arduino:avr:leonardo
Using board 'leonardo' from platform in folder: C:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo C:\Users\GG\AppData\Local\arduino\sketches\CEDB5789CFED8CCAFE1AEB67D28F75C3\sketch\ExtruderPSICONTROL.ino.cpp -o nul
Alternatives for functions.h: [CartersFunc@1.0.0]
ResolveLibrary(functions.h)
  -> candidates: [CartersFunc@1.0.0]
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc C:\Users\GG\AppData\Local\arduino\sketches\CEDB5789CFED8CCAFE1AEB67D28F75C3\sketch\ExtruderPSICONTROL.ino.cpp -o nul
Alternatives for SevSeg.h: [SevSeg@3.7.0]
ResolveLibrary(SevSeg.h)
  -> candidates: [SevSeg@3.7.0]
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg C:\Users\GG\AppData\Local\arduino\sketches\CEDB5789CFED8CCAFE1AEB67D28F75C3\sketch\ExtruderPSICONTROL.ino.cpp -o nul
Alternatives for EEPROM.h: [EEPROM@2.0]
ResolveLibrary(EEPROM.h)
  -> candidates: [EEPROM@2.0]
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\EEPROM\src C:\Users\GG\AppData\Local\arduino\sketches\CEDB5789CFED8CCAFE1AEB67D28F75C3\sketch\ExtruderPSICONTROL.ino.cpp -o nul
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\EEPROM\src C:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc\functions.cpp -o nul
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\EEPROM\src C:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg\SevSeg.cpp -o nul
Generating function prototypes...
C:\Users\GG\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\leonardo -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\CartersFunc -IC:\Users\GG\OneDrive\Documents\Arduino\libraries\SevSeg -IC:\Users\GG\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\EEPROM\src C:\Users\GG\AppData\Local\arduino\sketches\CEDB5789CFED8CCAFE1AEB67D28F75C3\sketch\ExtruderPSICONTROL.ino.cpp -o C:\Users\GG\AppData\Local\Temp\2596247215\sketch_merged.cpp
C:\Users\GG\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\GG\AppData\Local\Temp\2596247215\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\GG\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-IC:\\Users\\GG\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\GG\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\leonardo" "-IC:\\Users\\GG\\OneDrive\\Documents\\Arduino\\libraries\\CartersFunc" "-IC:\\Users\\GG\\OneDrive\\Documents\\Arduino\\libraries\\SevSeg" "-IC:\\Users\\GG\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\EEPROM\\src" "C:\\Users\\GG\\AppData\\Local\\arduino\\sketches\\CEDB5789CFED8CCAFE1AEB67D28F75C3\\sketch\\ExtruderPSICONTROL.ino.cpp" -o "C:\\Users\\GG\\AppData\\Local\\ [... cut ...]
### Expected behavior

Binaries should contain code other than only the bootloader.

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

Carter3DP, 2025-01-27:
I have found a workaround for this issue should anyone run across this.
Grab the board manager from [this forum post](https://forum.arduino.cc/t/new-atmega32u4-chip-uploaded-boot-loader-fine-but-serial-lights-does-not-work/1098032/12) and insert the link into 'Board manager URLs' in the Arduino IDE preferences. Then search and install "SparkFun AVR Boards" from the built in boards manager. Select the SparkFun Pro Micro as your board and compile, and now the binary should have your program included.
[Pastebin with working binary ](https://pastebin.com/H3qagzsK)

sterretjeToo, 2025-02-20:
As far as I can see this is a duplicate of https://github.com/arduino/ArduinoCore-avr/issues/411.

closed 2025-02-20 by per1234